ControlSystemStudio/phoebus #2922: v4.7.3 save and restore url setting

tynanford, 2024-01-27:
After upgrading the phoebus engine and the save and restore service to v4.7.3 there seem to be issues getting the phoebus client to connect to the service API.

Setting this variable - ```org.phoebus.applications.saveandrestore/jmasar.service.url=http://mydomain/save-restore``` in the settings.ini file doesn't impact the save restore client. And in the debugger I see it is always set to the default http://localhost:8080/save-restore.

https://github.com/ControlSystemStudio/phoebus/blob/v4.7.3/app/save-and-restore/app/src/main/resources/save_and_restore_preferences.properties

I think I was able to trace the issue back to this PR https://github.com/ControlSystemStudio/phoebus/pull/2816

georgweiss, 2024-01-28:
Due to refactoring of the code the settings is now:

```org.phoebus.applications.saveandrestore.**client**/jmasar.service.url=http://mydomain/save-restore```

Sorry for the inconvenience.

tynanford, 2024-01-29:
No worries, thanks for the help. I created a PR https://github.com/ControlSystemStudio/phoebus/pull/2927 to split out these `client` package settings so they show up correctly on the documentation pages

closed 2024-01-29 by tynanford