ControlSystemStudio/phoebus #2924: Alarm tree leaves should show PV information like other PV widgets

jembishop, 2024-01-29:
[image]

It would be useful to have relevant hover over information when you mouse over a leaf (or node) of the alarm tree, similar to the usual PV hover view in phoebus.

georgweiss, 2024-01-29:
The alarm clients consume data only from Kafka, they (intentionally?) do not connect to PVs like widgets do. 

In the context menu of a leaf node user may launch a PV aware view like Probe or Data Browser.

jembishop, 2024-01-29:
Would it be possible on hover to launch a PV subscription to make this work? I would volunteer to add this feature if you think its a good idea.

georgweiss, 2024-01-29:
In principle yes, but connection may take a bit of time so rendering data may come with a lag.

I do not mind, but I'd like @shroffk and @kasemir to comment.

kasemir, 2024-01-29:
Not sure how "on hover to launch a PV subscription" would work.
The tool tip is activated on the UI thread. You must provide the tool tip content right then. If you create the PV at that time, wait for it to connect, then show the text (and of course dispose the PV to avoid leaks), you're blocking the UI thread.
I'm OK with showing whatever information the alarm client has available at the time in alarm UI tool tips, but having the alarm client connect to the live PV sounds wrong. To get to the live PV, you can already open probe etc. from the alarm UI context menu.

jembishop, 2024-01-29:
Ok fair enough, thanks for your feedback. Looks like this feature might be more trouble than its worth so I will close the issue.

closed 2024-01-29 by jembishop