ControlSystemStudio/phoebus #2925: Change "CS-Studio" application title (and references) to "Phoebus"

jembishop, 2024-01-29:
I appreciate that there are historical reasons why the application title and references point to "CS Studio", however this situation its quite confusing to newcomers, especially when there is another repo called "cs-studio". Would it be acceptable if I made this change?

kasemir, 2024-01-29:
We regrettably lack a public relations department to better convey this, but the idea should be that it's very much CS Studio.

CS Studio started out with an Eclipse-based implementation (SWT, plugins/extension points, very complex build system, must use Eclipse IDE).
Now we have CS Studio in a Phoebus-based implementation (JFX, SPI, mvn or ant, may use Eclipse or vi or any other IDE).

The Eclipse-based code based is still used in several places and there are a few commits each year to fix bugs or adopt to changes in the JDK or Eclipse.
The Phoebus-based implementation, however, is by now the predominant one with continuing additions and improvements. So phoebus is not "CS-Studio, also". It's "_The_ current version of CS-Studio".

Just like EPICS moved from 68k CPUs/vxWorks and SunOS to many CPUs and Linux while remaining "EPICS".
Just like Mac OS moved from PPC via Intel to Apple Silicon and remained "Mac OS". Actually very similar to Mac OS, which has intermediate cute names like Jaguar,  El Capitan, Sonoma, "Phoebus" is like "Sonoma".

jembishop, 2024-01-29:
Thanks for your reply. I understand your reasoning here, but I would still argue that in this case a new name would be warranted. There are many cases of open source forks being renamed eg Postgres, Joomla, Ubuntu. And Phoebus comes with such an expanded feature set it would be worthy of a differentiating name in my opinion. 

The main problem with the current state of affairs is its confusing to have https://github.com/ControlSystemStudio/cs-studio and yet https://github.com/ControlSystemStudio/phoebus is meant to be the latest version of the latter, even with the disclaimer in the README. If Phoebus was simply a new version tag of the cs-studio repo, then this would be fine, but this is not the case. In your EPICS example https://github.com/epics-base/epics-base is the latest version, there is not another repo with a different name meant to be the latest version of epics-base, which would be confusing.

This is further complicated by facilities (such as ours) that retain both old cs-studio and phoebus because of some backwards compatibility reasons, and it would be nice for users to tell at a glance which is which. 

For newcomers to the project, I think simply having it under a different name is a lot more understandable, and the documentation can ensure full acknowledgment of the CS-Studio heritage. It would lead to a much cleaner brand image, which would encourage more people to use the tool.

kasemir, 2024-01-29:
At ICALEPCS 2017 I got chewed out for abandoning CS-Studio and instead pushing for a different new thing. So I have a personal interest in explaining that there's no intention at all to leave anybody stranded. Quite the opposite: The Eclipse-based code base was getting out of hand and becoming impossible to build. "Phoebus" was an attempt to keep "CS-Studio" viable, to run existing displays with minimal manual adjustments, to get the same data browser etc. 
But you're right, that failed, nobody gets that the idea was to keep CS-Studio going. It is CS-Studio = Eclipse, Phoebus = not Eclipse and the latter is seen by most as a different thing, so might as well stop calling it CS-Studio.

jembishop, 2024-01-29:
Thats fair enough at that point in time, however given it sounds like more and more people are interested in using Phoebus, in my opinion it would be best to prioritize understandibility for new users. By now it should be clear to any critics that backwards compatibility with CS-Studio is fully supported.

jembishop, 2024-01-30:
@kasemir Is it ok to raise a PR to implement this change? It sounds like you think this would be the right way forward from your last response.

kasemir, 2024-01-30:
Sure, whatever. While you're at it: Phoebus started in 2017. Every 10 years there's a new EPICS display tool, so in 3 years you'll have to come up with a new name and maybe also logo.

jembishop, 2024-01-30:
@kasemir Ok, forgive me but I can sense some reluctance on your part, I would only want to go ahead and implement this if the whole team including you would be in agreement. I can raise the pull request and there can be full discussion there with input from the other contributors, to further confirm if thats the direction the community wants to go. 

You mention "Every 10 years there's a new EPICS display tool", the reason I brought this up was to try and avoid such circumstances by improving the image of Phoebus so it can be identified as a mature production grade tool more easily. I don't think any more names or logos should be required as existing evolutions such as a web client can be named "phoebus-web" and can coexist with the other "phoebus-x" services which was not the case with the fork from cs-studio. The implication being that all these services have compatibility and integration with eachover. 

My intention is just for the tool to come off in the best light. I think that in the future there could be further improvements, for example putting some videos of controls screens on the README, adding docker compose configurations for the phoebus microservices stack and more. But this change would be quite easy and would increase clarity and reduce confusion in my view.

georgweiss, 2024-01-30:
@jembishop, I think I lean towards agreeing with @kasemir. At ESS we faced the same issue when 5 years ago we decided to go all in for Phoebus after using CSS Eclipse for some years. A lot of confusion and sometimes frustration. 

At ESS Phoebus has become synonymous with CS Studio simply because Phoebus is the only variant we use. However, it's not more than a year or so we converted the last "legacy" OPIs to Phoebus.

shroffk, 2024-01-30:
Thanks @jembishop for putting effort towards addressing the PR related to Phoebus.

The frustration with this topic has been because part of the collaboration has complained that we have abandoned them by giving up eclipse...so we decided to continue with CS-Studio and treat Phoebus as a named upgrade to underscore continuity... which has confused others.

I am Ok with us starting to replace CS-Studio with Phoebus in user facing places... we will have to come up with a new logo since the "C" based current logo will not make sense if the name is Phoebus.

> My intention is just for the tool to come off in the best light. I think that in the future there could be further improvements, for example putting some videos of controls screens on the README, [...]

Awesome!!! the optimism of youth is greatly needed :)

In the meanwhile, if you can think of ideas by which we can try to preserve CS-Studio to signal continuity and emphasis "Phoebus" as the latest and greatest version of CS-Studio that would be best

kasemir, 2024-01-30:
Looks like there is no obvious best choice. Simply need to vote. If the majority believes that continuity with CS-Studio is best expressed by no longer calling it CS-Studio, then I might not understand that, but I no longer care that deeply, either, because I'll get complaints either way.

georgweiss, 2024-01-31:
Now that things have settled at ESS I do not have any opinion on naming. However, we need to consider https://www.controlsystemstudio.org/

jembishop, 2024-01-31:
@shroffk Thanks for your feedback. If I understand correctly the fear from CS-Studio Eclipse users is coming from worry that their .opi control screens will not be compatible with Phoebus. In my option the best way to assuage these fears if this proposed rename happens would be to say in the the readme of the project something like:

>  … Phoebus started as a fork of CS-Studio
and we are and will always be fully committed to providing full backwards compatibility with the .opi control screens, and providing easy migration pathway from CS-Studio. 

And in the description

> Phoebus is a …. with full compatibility with  CS-Studio control screens.

I agree a new logo would be warranted, I imagine that would be in a different PR but the same release.

Addressing your comment here:

> In the meanwhile, if you can think of ideas by which we can try to preserve CS-Studio to signal continuity and add emphasis to "Phoebus" as the latest and greatest version of CS-Studio that would be best

If we want to go this route then I think simply changing the title from “CS-Studio” to “CS-Studio/Phoebus” would help.

Regarding your point @georgweiss about the website. From the layout of the project in GitHub, it gives the impression that the application “Phoebus” is being managed by the “ControlsSystemStudio” organisation. Similar to how Canonical is the organisation managing Ubuntu.

So to me it doesn’t seem that confusing that the website is named after the organisation. Maybe one day in the future it would be slightly more clearer to rename the organisation “phoebus-org”. Looks like the phoebus.org domain already links to the website anyway which will simplify that. But that is out of scope of this proposed PR anyway.

Any references on the website that could cause confusion would be need changed as part of this proposed PR though.

@kasemir Do you think a GitHub poll would be appropriate to implement your proposed vote?

kasemir, 2024-01-31:
GitHub poll is fine.

As for "full ... compatibility", I don't see how you want to guarantee that. My point has always been to explicitly NOT promise pixel-by-pixel equality. What we aim for is reasonably high compatibility, like "run existing displays with minimal manual adjustments" as mentioned above.
EDM, MEDM, Eclipse/SWT/BOY, Eclipse/JFX/DisplayBuilder, now Phoebus/JFX/DisplayBuilder and Display Builder Web Runtime are all totally different pieces of code. Still, you can open EDM, MEDM or BOY displays in the latest display builder, and display builder files are for the most part represented in the web runtime. As long as you simply added a widget and set the text or PV name, the result will be very similar. On the other hand, if you used every little feature of each widget, that will not translate.
See also https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility and https://controlssoftware.sns.ornl.gov/training/2022_USPAS/Presentations/03%20User%20Interfaces.pdf

closed 2024-04-28 by jembishop