ControlSystemStudio/phoebus #2938: Write of NTTable fails over pva gateway

georgweiss, 2024-02-06:
So we observe some interesting discrepancies between Phoebus and CLI pvput.

Writing NTTable (VTable) from Phoebus works fine if connecting directly to local IOC, but not over pva gateway. Packet analysis shows that if I comment out L166 in ```PutRequest```, the write over pva gateway succeeds, as does write to local IOC. However, write of for instance scalar PV fails.

The command line pvput succeeds in both cases (local and gateway).

In my specific example the indices to write are computed as ```{6,7,8}``` by Phoebus. Commenting out L166 instead results in ```{7,8}```, which is what pvput also seems to calculate.

So I'm wondering if the pva gateway should be suspected to not do what is expected. Same with pvput, why is the index array different to Phoebus?

@kasemir, any ideas?

```PuRequest``` has two calls to ```changed.set(data.getIndex(field));```, but that is not the issue here. Will fix that.

kasemir, 2024-02-06:
I'm afraid it's not clear to me from available documentation of the protocol what the "correct" way would be.
Which gateway are you using? The old one based on pvAccessCPP, or the new one based on PVXS?
The IOC's QSRV still uses pvAccessCPP, unless you build from a branch that supports PVXS.
So what you see is likely a different understanding implemented in pvAccessCPP vs. PVXS.
Not sure if it's possible to please both.

georgweiss, 2024-02-13:
Fixed in #2942