ControlSystemStudio/phoebus #2940: Alarm status info in tooltip does not update

zhangt58, 2024-02-07:
It seems the `HIHI/HIGH/LOW/LOLO` info in the tooltip does not update in Phoebus version 4.7.3-SNAPSHOT used at FRIB, a new bug?

zhangt58, 2024-02-08:
Yes.

kasemir, 2024-02-08:
Are you using `ca:` or `pva:`?

For `ca:`, do you enable  `org.phoebus.pv.ca/dbe_property_supported=true`? Default is `false` to support older CA servers.

For `pva:`, check if the client sees updates:

```
phoebus.sh -main org.epics.pva.client.PVAClientMain monitor NameOfPVWithoutAnyPrefix
```

kasemir, 2024-02-08:
With the PVA client I do see updates when HIHI/HIGH/LOW/LOLO are changed, but only if the associated HHSV/HSV/LSV/LLSV are configured.

zhangt58, 2024-02-08:
> Are you using `ca:` or `pva:`?
> 
> For `ca:`, do you enable `org.phoebus.pv.ca/dbe_property_supported=true`? Default is `false` to support older CA servers.
> 
> For `pva:`, check if the client sees updates:
> 
> ```
> phoebus.sh -main org.epics.pva.client.PVAClientMain monitor NameOfPVWithoutAnyPrefix
> ```

The PVs were with `ca:`, and `dbe_property_supported` defaulted as `false`. I just tried with `true`, and the tooltip did update. Thanks for your help!