ControlSystemStudio/phoebus #2951: Inconsistent/buggy behaviour with alarm tree server

jembishop, 2024-02-16:
Hi there

We have noticed that (at least in our case) the alarm tree can be quite buggy. For example sometimes we can reload the alarm tree page and it will have the wrong number of nodes. You can solve this by setting the alarm_tree_startup at a high value say 30s, but thats not ideal either.

Other times, I have noticed it can get in a bugged state and won't let you delete nodes (see gif below).

Also we have noticed that it can seem to replay the history of deletes and readds, unnecessarily slowing down the load times.

I'm not sure of the issue here, but one improvement that could definitely be made is that it looks like the kafka messages are replayed to generate the alarm tree state. A faster and for certain more reliable method, is to have an initial kafka message that simply contains the latest state, and it initializes based on that payload. This method would be a lot more reliable and faster than the current one for sure. There could also be a refresh button which can do this, incase your local state gets wrong somehow. This "batched" approach would also work well for making sure the commit history on the alarm-config-logger is minimized, as it looks like it makes a commit for every node you add when you do an import, which is not ideal. 

[image]

kasemir, 2024-02-16:
The alarm system use of Kafka is based on 'compacted' topics.
Some item is added, then removed -> Kafka log eventually only contains the removal.
Some item is OK, Minor, Major, Major-Ack, OK -> Kafka log eventually only contains the OK state.

If you see a long replay of history, the compaction isn't working. Check https://github.com/ControlSystemStudio/phoebus/blob/master/app/alarm/Readme.md:
Did you set `auto.create.topics.enable=false` and then create the topics with the script that configures them as compacted? At the bottom of the readme, verify that compaction is happening in the log cleaner.

jembishop, 2024-02-20:
Thanks very much @kasemir, setting this config option fixed the issues above for us. Apologies for not reading the README properly!

closed 2024-02-20 by jembishop