ControlSystemStudio/phoebus #2956: PvAccess - Access to a position of an array of a PV

JordiAnglada, 2024-02-22:
Hello all,

I have problem with Phoebus, when I try to link the pv's from my pva server to a widget I can't access to a position of an array.

I've been looking to other issues and I have seen other people have encountered the same problem, I've tested some of the suggested solutions and ensured that my array is at the end (pva://pva/subfield/subelement[5] is OK, pva://pva/subfield[4]/subelement is not) but non of them are working. 

Is this a bug on Phoebus ?

Maybe is something wrong with my server, here it is the server I'm using: https://github.com/JordiAnglada/PVAccess/blob/main/simple_pva_server.py

Issues: https://github.com/ControlSystemStudio/phoebus/issues/2657

Here you have some testing I've done trying some suggested fixes.

Only giving array path pva://db(takes by default the position [0] of the array, the value actually changes, but can't access to other positions than 0):  
[image]

Providing full pv path as suggested in other issues (not taking any value):
[image]

Testing with Channel Access (works fine):
[image]

kasemir, 2024-02-22:
With Channel Access, the element access is handled on the server side.
We just send "simple:C.VAL[2]" as a PV name, and the IOC then decodes that, picks the array element, and to us that looks like a scalar PV. You get the same result from any CA client because it's handled on the server.

For PVA, the syntax for handling arrays isn't generally defined, the implementation is all on the client side, and different for the legacy C++ lib, the new PVXS C++ lib, and what @shroffk implemented in the phoebus PV.
I'd say you have two options:

Don't use array access. Add subarray records to the IOC to pick elements, so you have scalar PVs.

Or work on implementing consistent PVA array handling in all implementations.

JordiAnglada, 2024-02-23:
Thanks a lot for your response.

For timing issues I think I will take the rout of using Channel Access.
But is nice to know that a PVA array handler needs to be implemented, for some future work.