ControlSystemStudio/phoebus #2957: openjdk version

caraxlr, 2024-02-22:
Hi there,

I was wondering how compatible openjdk 17 is with phoebus v4.7.3? I found out that phoebus v4.7.3 with openjdk17 seems to solve xy plot loading problem that phoebus 4.6.6 with openjdk11 has. Do you have any ideas of the cause?

[image]

Thanks a lot!

georgweiss, 2024-02-23:
My users have not reported issues with X/Y plot under Java 11. We have used Phoebus (on CentOS 7) extensively for about four years.

caraxlr, 2024-02-23:
Ok, thank you! And I wonder is there any compatibility issue of openjdk 17 with v4.7.3? I saw that with tag v4.7.3, the required openjdk of phoebus is still 11.

georgweiss, 2024-02-23:
We have done some testing of 4.7.3 under Java 17 and not seen any issues. But these tests were done done systematically -they were not exhaustive. 
The 4.7.3 build artefacts are Java 11 compatible, it would be surprising to see any issues when running 4.7.3 under Java 17.

caraxlr, 2024-02-23:
got it, thanks again!

closed 2024-02-23 by caraxlr