ControlSystemStudio/phoebus #2964: How to change macro value in run time from an action button or click.

JordiAnglada, 2024-02-28:
Hello all,

I want to change the vale of a macro from an action button, this macro is used to change the PV name on an embedded display (ca://LIPAC-MPS:EV_ST.VAL[$(N1)]).

The idea is by clicking different buttons the macro N1 updates and the embedded display shows different positions of a PV array.

I failed to change the macro value with this action button. There is any way to accomplish that ?

I already tried declaring the macro in a group and anchoring the action button in it (like CS Studio)

[image]

kasemir, 2024-02-28:
In general, macros are not supposed to change at runtime.

If you want to use an embedded display with a few different macro settings, look at the navigation tabs widget.

If you need to update macros of an embedded display in a more elaborate way, you'll find a script example in the example displays, specifically https://github.com/ControlSystemStudio/phoebus/blob/master/app/display/model/src/main/resources/examples/embedded/update_display.py

JordiAnglada, 2024-03-01:
Thanks for your response, 

I implemented the script and works fine!

closed 2024-03-01 by JordiAnglada