ControlSystemStudio/phoebus #2966: Application Menu Bar Hides When Releasing Mouse Click if no Window Manager or Rudimentary Tiling Window Managers (dwm) are Used

hydro-controls, 2024-03-01:
If Phoebus is opened directly on X (no window manager used), or if Phoebus is opened in very lightweight tiling window managers (suckless dwm), the application menu bar is effectively inaccessible. When left clicking an item on the menu bar, the initial menu is shown but no items can be selected. Additionally, when the mouse button is released, the menu item hides.

This issue is similar to but #640 (and items referenced within) but different from and is likely JavaFX related.

kasemir, 2024-03-01:
I've seen that with a small  https://www.yoctoproject.org type linux image, don't remember the exact minimalistic window manager that it used.

You can try GTK2 vs GTK3 (https://github.com/ControlSystemStudio/phoebus/wiki/JavaFX), but overall yes. Won't work well without a mouse, keyboard and monitor, either ;-)

hydro-controls, 2024-03-01:
I have had reasonable success with a stripped version of [openbox](https://github.com/Mikachu/openbox) so far, but am working to really lock down the human machine interface operators would use. Ideally, no window manager would be preferred for our implementation. Understand the wontfix label.

Mouse, keyboard, and monitor are all functioning satisfactorily.

kasemir, 2024-03-01:
The CS-Studio desktop product is meant to combine several tools. You can open multiple displays and arrange them in certain ways. You look at a display, then right-click on some text update and open the history for that PV in a data browser. Next you submit a screenshot to a logbook or email it.

If you want a stricter system where there's just a display page, consider the display builder web runtime,  https://github.com/ornl-epics/dbwr, and show that in a full-screen web browser.