ControlSystemStudio/phoebus #2970: Bug? open previous display button cannot be reused

caraxlr, 2024-03-07:
Hi,

We're trying to use the phoebus 4.7.3, however, we've noticed that the previous display button seems fail to function if it has multiple layers? The first click can bring me back to the previous one, but the second click seems to fail. 
[image]
Version 4.6.6 seems working fine. Thanks a lot!

caraxlr, 2024-03-08:
Here is a simplified example of this problem. 
[example.tar.gz](https://github.com/ControlSystemStudio/phoebus/files/14542878/example.tar.gz)
- `Main`.bob has `embedded1`.bob embedded
- `embedded1`.bob has a button to open display `display1`.bob
- `display1`.bob has `embedded2`.bob embedded
- `embedded2`.bob has a button to open display `display2`.bob

And the problem is:
- after two button click from `Main` → `display1` → `display2`
- back button only works from  `display2` to `display1`
- but using the dropdown can go back from `display1` to `Main`

Also the problem may be relevant to the macro usage, when I remove all the macros, the navigation seems working fine.

Thanks again!

caraxlr, 2024-03-13:
Here is one of the error:

2024-03-12 19:58:35 WARNING [org.phoebus.framework.macros] Incomplete macro expansion EVR='$(P)-$(SYS):EVR1'
java.lang.Exception: Macro spec EVR='EVR=$(P)-$(SYS):EVR1' does not fully resolve
	at org.phoebus.framework.macros.Macros.expandValues(Macros.java:263)
	at org.csstudio.display.builder.model.widgets.MacroWidget.expandMacros(MacroWidget.java:106)
	at org.csstudio.display.builder.model.widgets.GroupWidget.expandMacros(GroupWidget.java:256)
	at org.csstudio.display.builder.model.DisplayModel.expandMacros(DisplayModel.java:306)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.loadModel(DisplayRuntimeInstance.java:387)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.lambda$loadDisplayFile$4(DisplayRuntimeInstance.java:301)
	at org.phoebus.framework.jobs.Job.execute(Job.java:50)
	at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:54)
	at org.phoebus.framework.jobs.JobManager.lambda$schedule$0(JobManager.java:45)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:840)

I'm guessing this error could be relevant to the backward macro assignment (the relevant macro values may not be stored.)

georgweiss, 2024-03-14:
@abrahamwolk, @kasemir, would you be able to comment?

caraxlr, 2024-03-14:
Thank you! We're aiming to deploy the new version before our commissioning, which is scheduled within 3 weeks. Your prompt attention to solving this problem would be greatly appreciated. Thanks again!

caraxlr, 2024-07-25:
Hi there, I have some new findings for this ticket. It seems to be relevant to the order of macros. If I change macros from
      <macros>
        <EVM>$(P)-$(SYS):EVM1</EVM>
        <EVR>$(P)-$(SYS):EVR1</EVR>
        <P>MCR</P>
        <SYS>MT</SYS>
      </macros>
to 
      <macros>
        <P>MCR</P>
        <SYS>MT</SYS>
        <EVM>$(P)-$(SYS):EVM1</EVM>
        <EVR>$(P)-$(SYS):EVR1</EVR>
      </macros>

the back button works. And the problem occurs after version 4.7.2. There was only two commits in DisplayNavigation.java. I think the changes may be related to how the macros are processed? I'd appreciate it if you could give me some hints about what's possibly the changes, which files or commits are related, thanks again.

closed 2024-08-13 by caraxlr