ControlSystemStudio/phoebus #2977: JVM crash on Mac OS 14.3.1 (Sonoma) and Temurin 17.0.10+7

georgweiss, 2024-03-18:
See: https://github.com/adoptium/adoptium-support/issues/1043

Temurin 17.0.8+7 works, though, as does Temurin 21.0.1

georgweiss, 2024-03-19:
@kasemir, @shroffk, there is an interesting comment in the issue ticket I created on Adoptium...

kasemir, 2024-03-19:
"JFX19 requires JDK19"??

https://github.com/openjdk/jfx/blob/jfx21/doc-files/release-notes-19.md states "JavaFX 19 requires JDK 11 or later".

https://github.com/openjdk/jfx/blob/jfx21/doc-files/release-notes-21.md states "JavaFX 21 requires JDK 17 or later", and similar statements were in previous releases, so there is generally no exact version match required.

georgweiss, 2024-03-19:
Told you the comment is interesting...
Will update with your input.

shroffk, 2024-03-19:
wtf

georgweiss, 2024-03-26:
Bug submitted also to OpenJDK: https://bugs.java.com/bugdatabase/view_bug?bug_id=JDK-8329016