ControlSystemStudio/phoebus #2979: Long scrolling text box

tanvi-ashwarya, 2024-03-18:
There's an issue with long scrolling text box where whenever the input PV updates, the text box automatically scrolls to the top.

kasemir, 2024-03-18:
If you want it to scroll to the bottom, set vertical alignment to bottom.

tanviash, 2024-03-20:
@kasemir This does keep the text at the bottom. However, it does not hold the current line number when the PV updates. Every time it updates, it snaps to the bottom.

kasemir, 2024-05-21:
It might be hard to predict what the user prefers: Keep the first line, last line, or the scroll position. In practice, whatever you do is likely wrong with 66% of the people.

The comments in https://github.com/ControlSystemStudio/phoebus/blob/3596313aae03b04a14ed191d1b9714555f750850/app/display/representation-javafx/src/main/java/org/csstudio/display/builder/representation/javafx/widgets/TextUpdateRepresentation.java#L287 also hint to problems at least in the past with trying to control the scroll position.

Still, I think I found something that at least seems to work on a Mac:
If the "Vertical Alignment" of a text entry or text update widget is set to "Middle", then we'll try to keep the scroll position with value changes.
That way you can create the display with either "Top" or "Middle" to get the scroll to the top as before, or keep the scroll position.