ControlSystemStudio/phoebus #2980: Editing multi-line text boxes

tanvi-ashwarya, 2024-03-18:
Editing multi-line text boxes doesn’t handle newlines when editing “in-place” (double click on the text box to edit in the main editor view).  Newlines are removed and cannot be recovered.

On display:
 
[image]

Editor:
 
[image]

Editor after double-click to change the text, newlines are immediately lost, and it is updated without newlines.
 
[image]

[image]

kasemir, 2024-03-18:
There's a related issue with the inline editor: I keep getting it into a state where it removes spaces.
Take a label that contained "Hello". Use the inline editor to change that to "Hello    world" and the result becomes "Helloworld". Not with a freshly started instance of CS-Studio, but after running for a while. No idea what makes it get into that state. When fixing the handling of newlines, maybe something obvious comes up that would explain the removal of spaces.

kasemir, 2024-03-18:
For what it's worth, I doubt the inline editing of multi-line text was ever fully supported because typing 'Enter' would save the current text, closing the inline editor. So you could tweak the text but not add new newlines. For full editing, you always had to use the "..." button next to the text property to open a full dialog.
Still, it's reasonable to have the inline editor show the text with newlines and then allow tweaking the text. Or maybe show newlines as '\n' and convert them back when saved.

kasemir, 2024-03-18:
The inline editor is a TextField, not a TextArea. So it was always limited to one line.
A quick workaround would be showing newlines as `\n` and replacing them back to newlines.
This does allow editing the text, adding/removing newlines, but it looks different than showing actual newlines.
For full editing, the "..." button next to the text property would still be best.

tanviash, 2024-03-18:
Closing the ticket, fix provided by Kay.

closed 2024-03-18 by tanviash