ControlSystemStudio/phoebus #2990: Some URLs are not displayed with action button (Open Web Page) and Web Browser widgets

Mathis-Hu, 2024-03-26:
When entering some https:// URLs, the page is not loaded, it seems to happen when the page requires an authentication to access.

Some of my users want to access the web page of a web server with instrumentations, but they cannot with both widgets.

For the Open Web Page action (responsible class : ModelResourceUtil.java) , it seem that the code is checking if the page exist and is trying to read it. So if the page is not correct or have an authentication, it will not load it, and you'll not have any pop up to inform you that it doesn't work.
I think it would be better if Phoebus only check if the page exist before opening it. Then if the URL is not correct, it will load anyways and the user can see that it's not correct, and it will allow the users to use URL for other cases (like the use case of my users).

Our lab modified the class so now phoebus only check if the page exist, and it work for our users. I will make a PR for that if it does not pose security issue or anything that justified the original operation.

For the web browser widget, it seems that phoebus uses WebEngine from javafx and that it's that class that manage the loading of web page. Do you think it would be possible to change some code to make the Web Browser less restrictive with URL like the changes I made for the Open Web Page action ?

@kasemir

kasemir, 2024-03-26:
For the "Open Web Page" action, we try to simply pass the URL on to the system web browser (Firefox, Safari, Chrome, ...).
The ModelResourceUtil performs some additional gymnastics because we generally support relative paths etc., so it's checking a few options. Maybe the "Open Web Page" action can circumvent that, only expand macros and then pass the url on to the system web browser? Or reducing the try-to-read check to check-if-exists also sounds reasonable.
For the web browser widget, I see that mostly as a way to show formatted but otherwise static text, for example usage information that would he hard to duplicate with just labels. For anything beyond that, a designated web browser is better (more features, lower CPU load). But you're welcome to take over maintenance of the web browser widget and make it fully support anything anybody will ever try to accomplish with it.

Mathis-Hu, 2024-03-26:
Ok, thank you for your answer, I will make a PR with the commit for the Open Web Page action, for the Web Browser widget, I'll try to take a look at what could be done

tanviash, 2024-04-18:
We are seeing a similar problem when action 'Open Webpage' is called. 
[image]
[image]

kasemir, 2024-04-18:
#2991 should fix that, but it changes the flow beyond just a replacement of
```
final InputStream stream = openURL(resource_name);
stream.close();
return true;
```
in the `ModelResourceUtil` with 
```
URL url = new URL(resource_name);
url.openConnection();
return true;
```

Is your https:// problem fixed by just that replacement? Then let's do that.

shroffk, 2024-08-01:
@Mathis-Hu can you confirm is the above works?

Mathis-Hu, 2024-08-02:
@shroffk For the action open web page with action button it is working. For the Web browser widget I have a doubt, but I don't think I patched it.

katysaintin, 2025-01-29:
Hello @shroffk and @kasemir .
Could we do something for this issue ?
It's now OK for https:// in Open Web Page Action.

But in some case, we have securized page https:// and we have to enter a password, or select a certificate.
The problem is , Action Open Web page try to read the contents . Why ?

Why don't we let the external Browser manage the error ?

I think, that it's necessary to read the contents only for the WebBrowser widget.
Also, instead of do nothing, can we display the Error in the content instead of do nothing ?

If you are OK, I can review this PR and improve it.
Thank you for your help.

Katy

kasemir, 2025-01-29:
What about my suggestion from April 18?

Simply replace 
```
final InputStream stream = openURL(resource_name);
stream.close();
return true;
```
in the ModelResourceUtil with
```
URL url = new URL(resource_name);
url.openConnection();
return true;
```

Looks like that works for the open web page action, so let's do that.

katysaintin, 2025-01-29:
Hello @kasemir  Yep sound perfect for me !

I found a solution also for WebBrowser widget.  I'm using WebEngine.load(String) ; when a URL is not working. Here is a screenshot of my modifications.

In the following screenshot , I display errors on Forbidden access (Error 403) and Unknown Host . I think it is better than display nothing. Are you OK if I make a PR including your suggestion and my WebBrowser improvments ?

[image]

kasemir, 2025-01-29:
Yes. Displaying the error in the web widget as you did is good!

katysaintin, 2025-01-30:
> We are seeing a similar problem when action 'Open Webpage' is called. [image] [image]
> 
> One interesting observation is if we change https:// to http://, works okay but now with https://

@tanviash  PR #3246 will solved you issue
Best regards,

Katy

kasemir, 2025-07-28:
I assume this was a self-signed, locally created certificate?
Yes, those are troublesome. You need to import them into your JRE. Next, your IT support updates the JRE because of some security concern, so you need to import the same custom certificate again. At our site, we've given up on that. All certificates are now properly signed, eliminating the need for custom cert imports.