ControlSystemStudio/phoebus #3002: Resource option considers extensions

georgweiss, 2024-04-26:
Documentation states the PV table may be stated upon launch of Phoebus if command line specifies: ```... -resource pv://?myPvName```.

But this does not work, seems only extensions are considered to locate a matching application. Was that possible in legacy Eclipse-based CSS and lost in the transition.

In my view it would be nice to support application matching through scheme on top of matching on extension. I have a use case like ```... -resource saveandrestore://filter?name=foo``` in mind.

kasemir, 2024-04-26:
??
Your examples miss the app hint.
From "-help":
```
-resource 'file:/abs/path/file?app=display_runtime&MACRO1=value+1&MACRO2=xy' - Opens file with 'display_runtime' app...
-resource 'pv://?sim://sine&app=probe'                                       - Opens the 'sim://sine' PV with 'probe'.
-resource 'pv://?Fred&sim://sine&app=pv_table'                               - Opens two PVs PV with 'pv_table'.
```

georgweiss, 2024-04-26:
>Your examples miss the app hint.

So they do. Thanks.

closed 2024-04-26 by georgweiss