ControlSystemStudio/phoebus #3003: Open resource from command line fails on Mac OS

georgweiss, 2024-04-29:
Seen on Mac OS 14.4.1 (Sonoma):

- Open a display ```foo.bob``` in runtime mode
- Shut down Phoebus (i.e. memento will include the opi for restore upon restart)
- Launch Phoebus with ```-resource file:/path/to/foo.bob´´´
- In most attempts an exception is thrown:

```
java.lang.IllegalStateException: Attempted arrival of unregistered party for java.util.concurrent.Phaser@bf00549[phase = 0 parties = 1 arrived = 1]
	at java.base/java.util.concurrent.Phaser.doArrive(Phaser.java:383)
	at java.base/java.util.concurrent.Phaser.arrive(Phaser.java:624)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.awaitRepresentation(ToolkitRepresentation.java:384)
``` 

In those cases when exception is thrown I notice that ```ToolkitRepresentation``` calls ```phaser = new Phaser(1);``` twice before  ```phaser.arrive()``` is called the first time.

Not reproduced on Linux, Windows not tested. Reproduced both with Java 11, 17, 21 + JavaFX 19, as well as Java 17 and JavaFX 21.

mellguth2, 2025-02-17:
I have noticed the same problem under Windows 11 with openjdk 17 when using the "-resource ..." command line argument, i.e. an exception about "Attempted arrival of unregistered party..." was thrown and the "*.bob" document specified via "-resource ..." closes after pressing "OK" in the exception message box.

mdavidsaver, 2025-10-26:
A colleague running OSX encountered this error while running a 5.0.2 build downloaded from https://www.controlsystemstudio.org/download/

```
2025-10-25 16:30:43 SEVERE [org.csstudio.display.builder.runtime] Cannot load model from /Users/eric/src/Osprey/src/MPEX_ACQ/EPICS/opi/mpexACQ_sysmon.bob
java.lang.IllegalStateException: Attempted arrival of unregistered party for java.util.concurrent.Phaser@79cafd86[phase = 0 parties = 1 arrived = 1]
	at java.base/java.util.concurrent.Phaser.doArrive(Phaser.java:389)
	at java.base/java.util.concurrent.Phaser.arrive(Phaser.java:630)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.awaitRepresentation(ToolkitRepresentation.java:383)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.lambda$loadDisplayFile$5(DisplayRuntimeInstance.java:340)
	at org.phoebus.framework.jobs.Job.execute(Job.java:50)
	at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:54)
	at org.phoebus.framework.jobs.JobManager.lambda$schedule$0(JobManager.java:45)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:317)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
	at java.base/java.lang.Thread.run(Thread.java:1583)

```

mdavidsaver, 2025-10-26:
Both this occurrence, and #3607, may be seen in this log [this log](https://github.com/user-attachments/files/23145690/phoebus.txt).