ControlSystemStudio/phoebus #3006: SPI for widget interaction?

daykin, 2024-05-01:
Hi,

I would like to create a service that broadly logs when a particular widget is interacted with - say, clicking any button, changing the value of any text input in a DisplayBuilder screen. Is there a general way to capture these events? Eventually I would like to use these interactions to build a Neo4j graph database.

georgweiss, 2024-05-02:
Can you please elaborate a bit on this? Is the "context" of the widget (e.g. screen id) of interest? Is the purpose to track value - i.e. PV value - changes or any type of changes triggered by the widgets?

daykin, 2024-05-02:
This is still very whiteboard-level, but a couple of examples:  For a button, I'd want the screen ID (file name/URI works), widget ID and the actions associated with that button. Assuming it's an 'open display' action from screen X to screen Y, the graph database would reflect a node for X and Y, with an 'opened' relate between the two.

For another control e.g. text input, it would have screen ID, widget ID, PV name. The graph database would have a node for the screen ID, and a node for the PV, with a 'changed' relate. I don't really care what the exact value is.

kasemir, 2024-05-02:
Without some more defined use case, this sounds like an open-ended request which could result in a lot of work, code, overhead but few benefits.

For EPICS, you can install put-logging tools into the IOC which will log all write access, https://github.com/epics-modules/caPutLog/blob/master/docs/index.rst  With that information you can for example monitor what operators adjust most frequently during routine operation of your machine, and look for ways to automate that part of the control system to ease their load.

daykin, 2024-05-02:
We have caPutLog already, but it's only one piece of the puzzle. The complaint from operators is that the screens themselves are not well-integrated across systems. The design language is different, and there's settings all over the place. 

In the ideal world where this could be pulled off cleanly, we'd quietly be performing embedded time and motion studies without anything outwardly different to the end-user, via a modular component that only subscribes to existing endpoints (no hard dependencies in existing code). This would create 1) a graph database of human interactions, and 2) 2D kernel-density heatmaps overlaid on commonly-used displays, saved in a .png.

kasemir, 2024-05-02:
Looks like you want to hire a UX consultant who equips your operators with pupil trackers to see how long they look at what etc.

Check the JavaFX documentation, https://docs.oracle.com/javase/8/javase-clienttechnologies.htm, specifically event handling, https://docs.oracle.com/javase/8/javafx/events-tutorial/events.htm#JFXED117
There's a distinction between event filters and handlers. You can basically add an event filter to the top level pane, and it will see all events (mouse clicks, entered text, ..)
Not sure what's practical to add for this side project without impacting normal operational use.

daykin, 2024-05-03:
> Looks like you want to hire a UX consultant who equips your operators with pupil trackers to see how long they look at what etc.

Perhaps I buried the lede when I forgot to mention: one of the requirements is for the operators to not hate me.

daykin, 2024-05-03:
At any rate, I think the top-level interaction and WidgetRuntimes SPI are enough to get started. Thank you for the advice :)

closed 2024-05-03 by daykin