ControlSystemStudio/phoebus #3016: bug? medm converter does not convert a boolean expression correctly..

cpchuang, 2024-05-14:
There might be a bug in .adl file converter...

This part of code
https://github.com/epics-modules/motor/blob/master/motorApp/op/adl/motorx.adl
                calc="A!=0||B==7"
		chan="$(P)$(M).SET"
		chanB="$(P)$(M).STAT"

got converted to 
https://github.com/epics-modules/motor/blob/master/motorApp/op/bob/autoconvert/motorx.bob
      <rule name="vis_calc" prop_id="visible" out_exp="false">
        <exp bool_exp="!(pv0!==0||pv1==7)">
          <value>false</value>
        </exp>
        <pv_name>$(P)$(M).SET</pv_name>
        <pv_name>$(P)$(M).STAT</pv_name>

and cause syntax error
==============================================
java.lang.Exception: Cannot compile rule: rectangle:rectangle #40:vis_calc.rule.py
.
.
Caused by: SyntaxError: ("no viable alternative at input '=='", ('rectangle:rectangle #40:vis_calc.rule.py', 27, 18, '    if  not (pv0!===0 or pv1==7):\n'))
==============================================

should be "pv0!=0", not "pv0!==0".

Many display in motor module got affected.

prjemian, 2024-05-14:
Version info:

```bash
(base) s1idtest@kurtag ~/.../instrument/devices $ 2024-05-14 16:55:35 CONFIG [org.phoebus.product.Launcher] Loading settings from /net/s1dserv/APSshare/linux/x86_64/phoebus-4.7.3-SNAPSHOT/settings.ini
2024-05-14 16:55:35 INFO [org.phoebus.product.Launcher] Phoebus (PID 433646)
2024-05-14 16:55:35 WARNING [javafx] Unsupported JavaFX configuration: classes were loaded from 'unnamed module @20a6d62a'
2024-05-14 16:55:37 INFO [org.phoebus.channelfinder.ChannelFinderClient] Creating a channelfinder client to : http://localhost:8080/ChannelFinder
2024-05-14 16:55:37 INFO [org.phoebus.applications.alarm.logging.ui.AlarmLogTableApp] Creating a alarm logging rest client to : http://localhost:9000

(base) s1idtest@kurtag ~/.../instrument/devices $ more /etc/os-release
NAME="Red Hat Enterprise Linux"
VERSION="9.2 (Plow)"
...
```

prjemian, 2024-05-14:
The phoebus version is from

```bash
drwxr-xr-x  4 zzzzz zzzzz 4.0K Feb  6 13:12 phoebus-4.7.3-SNAPSHOT/
```

kasemir, 2024-05-21:
Thanks for the example, I can reproduce the issue and will look into it.

kasemir, 2024-05-21:
It now turns that expression into `!(pv0!=0||pv1==7)` which is valid.
If you fetch a binary from https://controlssoftware.sns.ornl.gov/css_phoebus/, check tomorrow for an update that includes this fix.

PR #3022 (2024-05-21): MEDM converter: Fix patching of bool expression
Fixes #3016
closed 2024-05-21 by kasemir