ControlSystemStudio/phoebus #3025: Initial parameter for executable construction

XavSPM, 2024-05-24:
Hello,

I am currently testing the possibility of creating executables with `jpackage`. So far, I have managed to create an RPM package and it works well.

However, I am looking to incorporate initial parameters into this package to match our installation. The idea is to have our default parameters without using the -settings option. What I want to do is that if there is no `/.phoebus` folder in the user's directory, the default parameters are those included in the package. I don't know if I can place a `settings.xml` or `settings.ini` file somewhere to achieve this.

Do you have any ideas on how to accomplish this? If this were possible, it would greatly help with the automation of creating executables according to the types and needs of users (operator, researcher, internal networks, external networks, etc.). I could thus include the pages (.bob) in the package and modify variables like these:
```
org.phoebus.ui/top_resources=$(phoebus.install)/interface/main.bob?app=display_runtime,Accueil
org.phoebus.ui/home_display=$(phoebus.install)/interface/main.bob?app=display_runtime
```

Additionally, is it possible to change the variable `phoebus.user = /home/xgoiziou/.phoebus` to:

`phoebus.user = /home/xgoiziou/.<new_name>` in the configuration file?

kasemir, 2024-05-24:
If all else fails, check the documentation, https://control-system-studio.readthedocs.io/en/latest/preferences.html 

_When you package phoebus for distribution at your site, you can also place a file settings.ini in the installation location (see [Locations](https://control-system-studio.readthedocs.io/en/latest/locations.html#locations)). At startup, Phoebus will automatically load the file settings.ini from the installation location, eliminating the need for your users to add the -settings .. command line option._

For. an example, see https://github.com/kasemir/phoebus-sns/tree/master/product-sns and its `settings.ini` file.

XavSPM, 2024-05-25:
Thank you @kasemir,

I had not properly read that part of the documentation. I tested with settings.ini, and the parameters are correctly taken into account.

However, I added the option `-Dphoebus.user=.SPR` to `jpackage` during the creation of the `rpm` package, and I did not get the desired result.

Without this option, the `phoebus.user` variable has the value: `/home/xgoiziou@arronax.com/.phoebus`.

With this option, the `phoebus.user` variable has the value: `.SPR`.

So, the folder is not located in the user's home directory.

Please note that I am launching the program with `/opt/spr/bin/SPR`.

I suppose I am doing something wrong, but I am not particularly comfortable with Java.

georgweiss, 2024-05-27:
@XavSPM , for what it's worth: we are using jpackage for Win and Mac. I have not been able to figure out how to build packages using a variable like ```-Dphoebus.user=$HOME/.SPR```. But my trials were conducted some time ago using Java14, cannot say if jpackage has changed since.

XavSPM, 2024-05-27:
Hello @georgweiss,

I tested your suggestion `-Dphoebus.user=$HOME/.SPR`. This effectively changes the`phoebus.user` variable to `/home/xgoiziou@arronax.com/.SPR`. However, when I install the program on another computer with another user, the variable remains equal to `/home/xgoiziou@arronax.com/.SPR`.

I think that when the package is created, the variable `-Dphoebus.user=$HOME/.SPR` becomes `-Dphoebus.user=/home/xgoiziou@arronax.com/.SPR` and therefore remains hard-coded in the program.

XavSPM, 2024-05-27:
By examining the code, I believe I have identified why it is not working.

In the Location.java file, we have this:

```java

    private static void initUser()
    {
        String phoebus_user = System.getProperty(PHOEBUS_USER);
        if (phoebus_user == null)
        {
            phoebus_user = new File(System.getProperty("user.home"), ".phoebus").getAbsolutePath();
            System.setProperty(PHOEBUS_USER, phoebus_user);
        }
    }
```

A solution could be to create a new variable called NAME_APP:

```java

    /** system property for phoebus installation path */
    public static final String PHOEBUS_INSTALL = "phoebus.install";
    /** system property for phoebus logged user */
    public static final String PHOEBUS_USER = "phoebus.user";
    
    public static final String NAME_APP = "name.app";

```
And modify the initUser method as follows:

```java

    private static void initUser()
    {
        String phoebus_user = System.getProperty(PHOEBUS_USER);
        String name_app = System.getProperty(NAME_APP);
        if (phoebus_user == null)
        {
            if (name_app == null) {
                name_app = ".phoebus";
            }
            phoebus_user = new File(System.getProperty("user.home"), name_app).getAbsolutePath();
            System.setProperty(PHOEBUS_USER, phoebus_user);
        }
    }
```
What do you think about this? Could it be useful for the users?

XavSPM, 2024-05-27:
> En utilisant `-Dphoebus.user=$HOME/.SPR` va s'étendre `$HOME`au moment de la construction. On peut éviter cela en échappant : `-Dphoebus.user=\$HOME/.SPR`. Mais Phoebus n'évalue pas la variable système `$HOME` lorsque l'emplacement des paramètres est déterminé.

Unfortunately, this doesn't work.

XavSPM, 2024-05-27:
> @georgweiss
> 
> By examining the code, I believe I have identified why it is not working.
> 
> In the Location.java file, we have this:
> 
> ```java
>     private static void initUser()
>     {
>         String phoebus_user = System.getProperty(PHOEBUS_USER);
>         if (phoebus_user == null)
>         {
>             phoebus_user = new File(System.getProperty("user.home"), ".phoebus").getAbsolutePath();
>             System.setProperty(PHOEBUS_USER, phoebus_user);
>         }
>     }
> ```
> 
> A solution could be to create a new variable called NAME_APP:
> 
> ```java
>     /** system property for phoebus installation path */
>     public static final String PHOEBUS_INSTALL = "phoebus.install";
>     /** system property for phoebus logged user */
>     public static final String PHOEBUS_USER = "phoebus.user";
>     
>     public static final String NAME_APP = "name.app";
> ```
> 
> And modify the initUser method as follows:
> 
> ```java
>     private static void initUser()
>     {
>         String phoebus_user = System.getProperty(PHOEBUS_USER);
>         String name_app = System.getProperty(NAME_APP);
>         if (phoebus_user == null)
>         {
>             if (name_app == null) {
>                 name_app = ".phoebus";
>             }
>             phoebus_user = new File(System.getProperty("user.home"), name_app).getAbsolutePath();
>             System.setProperty(PHOEBUS_USER, phoebus_user);
>         }
>     }
> ```
> 
> What do you think about this? Could it be useful for the users?

I tested by modifying this file and replacing the `name.app` variable with `-Dname.app`, and obtained the desired result.

With `java -Dname.app=.SPR_TEST -jar product-4.7.3.jar`, the `phoebus.user` variable is correctly set to `/home/xgoiziou@arronax.com/.SPR_TEST`.

And with `java -jar product-4.7.3.jar`, the `phoebus.user` variable is set to `/home/xgoiziou@arronax.com/.phoebus`.

XavSPM, 2024-05-27:
> In my jpackage script I have: ` --java-options -Dphoebus.user=\$HOME/.phoebus` Running Phoebus from the installed package will set `phoebus.user=$HOME/.phoebus`, see screen shot. What is missing is Phoebus evaluating `$HOME/.phoebus` in the same manner as a an interactive shell does, i.e. resolve `$HOME` to user's home directory. The Java API `System.getenv("$HOME")` should do the trick.
> 
> ![Screenshot 2024-05-27 at 13 13 44](https://private-user-images.githubusercontent.com/35602960/334080381-ba327a08-edb0-4f74-87f8-2966ea42290a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4MTQ5NjksIm5iZiI6MTcxNjgxNDY2OSwicGF0aCI6Ii8zNTYwMjk2MC8zMzQwODAzODEtYmEzMjdhMDgtZWRiMC00Zjc0LTg3ZjgtMjk2NmVhNDIyOTBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI3VDEyNTc0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZkNTVkZGM5N2E3YzI3Zjc1YWM2YTA3MDVmNzM2ODJlOWRlNzc0MTE0MWM1ZWMyNzRlNTE3Yjk1YjVlMDY3MjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7EW59i-rwQoD0rfaYYQDHExNJd5NWrT59-HMW5LrsUs)

I have tried all possible combinations with `System.getenv()`, but it doesn't change anything. The value of `phoebus.user` passed with the `-D` option is always interpreted as a string.

I have made a pull request with the program that works for me. #3028

georgweiss, 2024-05-27:
Sorry, I was wrong, the way I am able to resolve user's home directory (on Mac) is ```System.getenv("HOME")``` (no dollar char).

XavSPM, 2024-05-27:
Ah,

For me, on Linux (Fedora), this does not work. Is the command you are executing for this as follows?:
```
java -Dphoebus.user=System.getenv("HOME")/.SPR_TEST -jar product-4.7.3.jar
```

Or did you compile with jpackage?

Thank you again for your help.

georgweiss, 2024-05-27:
> ```java -Dphoebus.user=System.getenv("HOME")/.SPR_TEST -jar product-4.7.3.jar```

This will never work, ```System.getenv("HOME")``` is Java side code, e.g. in ```Locations.java```.

XavSPM, 2024-05-28:
Maybe this isn't really useful, but I thought that, since the changes aren't too great, it might be interesting for deploying dedicated applications.

closed 2024-06-07 by XavSPM