ControlSystemStudio/phoebus #3033: Alarm-logger - Kafka connection - shutdown_client

GDH-ISIS, 2024-06-03:
I am looking for advice for the alarm logger (compiled 31 May 2024) - we have a system running as docker containers whereby the alarm-logger encounters an exception (roughly every day). System starts fine, alarms appearing as expected - three topics (Accelerator, Upper, Lower manually configured in Kafka via script with auto.create=false). However after a period of time, the alarm-logger container seems to stop with an exception message (it sometimes seems as though this might coincide with some of our EPICS IOCs restarting, although I have no definitive evidence of this - still investigating). Note - on restarting only the alarm logger, the services appear to catch-up and the alarms appear in the Windows phoebus client as expected.

Sorry to ask, but I have been investigating the configurable parameters for some time without a resolution. Any advice much appreciated.

GDH-ISIS, 2024-06-03:
2024-05-31 17:35:08 INFO [org.phoebus.alarm.logging.AlarmLoggingService] Alarm Logging Service (PID 1)
Commands:
	help             - Show help.
	shutdown         - Shut alarm logger down and exit.

 > Jun 01, 2024 1:04:24 PM org.apache.kafka.streams.KafkaStreams handleStreamsUncaughtException
SEVERE: stream-client [streams-Accelerator-alarm-messages-6a72af80-55e9-4d7d-a504-a7b26dc83572] Encountered the following exception during processing and the registered exception handler opted to SHUTDOWN_CLIENT. The streams client is going to shut down now. 
org.apache.kafka.streams.errors.StreamsException: org.apache.kafka.common.KafkaException: Encountered corrupt message when fetching offset 2135773 for topic-partition Accelerator-0
	at org.apache.kafka.streams.processor.internals.StreamThread.runLoop(StreamThread.java:657)
	at org.apache.kafka.streams.processor.internals.StreamThread.run(StreamThread.java:579)
Caused by: org.apache.kafka.common.KafkaException: Encountered corrupt message when fetching offset 2135773 for topic-partition Accelerator-0
	at org.apache.kafka.clients.consumer.internals.AbstractFetch.handleInitializeCompletedFetchErrors(AbstractFetch.java:641)
	at org.apache.kafka.clients.consumer.internals.AbstractFetch.initializeCompletedFetch(AbstractFetch.java:514)
	at org.apache.kafka.clients.consumer.internals.AbstractFetch.collectFetch(AbstractFetch.java:283)
	at org.apache.kafka.clients.consumer.KafkaConsumer.pollForFetches(KafkaConsumer.java:1262)
	at org.apache.kafka.clients.consumer.KafkaConsumer.poll(KafkaConsumer.java:1186)
	at org.apache.kafka.clients.consumer.KafkaConsumer.poll(KafkaConsumer.java:1159)
	at org.apache.kafka.streams.processor.internals.StreamThread.pollRequests(StreamThread.java:1014)
	at org.apache.kafka.streams.processor.internals.StreamThread.pollPhase(StreamThread.java:962)
	at org.apache.kafka.streams.processor.internals.StreamThread.runOnce(StreamThread.java:766)
	at org.apache.kafka.streams.processor.internals.StreamThread.runLoop(StreamThread.java:617)
	... 1 more

GDH-ISIS, 2024-06-17:
Will investigate further and report back.

jeonghanlee, 2024-08-16:
@GDH-ISIS 

We identified the similar issues a long ago, after several months investigation, we have that solution on our specific case. You may try to test the latest pull request https://github.com/ControlSystemStudio/phoebus/pull/3111

GDH-ISIS, 2024-10-21:
Since commit ff78e8a30b7c02c79356ca85a68950f5d094ce88 or #3136 (we re-deployed on 16th Sept - again using docker), our circumstances have much improved. The alarm-logger service now appears stable and as far as I am aware all alarm events have been passed without any problems. No shutdown events recorded and it looks like our historical logs are complete. I think this can now be closed. Happy days 

Many thanks for all the effort. G

closed 2024-10-21 by GDH-ISIS