ControlSystemStudio/phoebus #3040: Stop persisting preferences in `.phoebus.userPrefs`

kasemir, 2024-06-12:
Preference settings start with the `xxx_preferences.properties` files for build-in settings, then adds persisted values, and finally values provided via `-settings /path/to/file` on the command line.

The end result is then persisted in `$HOME/.phoebus.userPrefs` (default, can be changed).
This persistence was done to mimic Eclipse. In Eclipse, a preferences GUI allowed runtime changes, which then update the persisted settings and get loaded on restart.

For us, the persisted preferences pose a few problems:

If you start a program with `-settings /path/to/settings`, those values get persisted.
If you next start the program without `-settings...`, you still get what's in `/path/to/settings` because that got persisted.

This includes the case where an archive tool might have been started with `org.phoebus.pv.ca/monitor_mask=ARCHIVE`. When the same user next starts the phoebus GUI without specifically settings the monitor mask, you still run with ARCHIVE and get lower update rates.

The `org.phoebus.framework.preferences.FileSystemPreferences` used to persist preferences is copied from the JDK (not public API) and extended to check  `$HOME/.phoebus.userPrefs`. It includes deprecated API, so will eventually need an update.

Suggestion is to simply remove the preference persistence, and only use
1) `xxx_preferences.properties` files for build-in settings
2) `-settings /path/to/file` on the command line

tynanford, 2024-06-12:
+1, I also think the persistence of preference settings causes more confusion than help at our site

kasemir, 2024-06-12:
Started to look at an implementation.

We do use the `java.util.prefs.Preferences` API. But instead of the default implementation which stores the prefs in some Mac OS way, some Linux files, or not sure what it uses for Windows, we copied the Linux `FileSystemPreferences` and updated it to store the prefs under `$HOME/.phoebus/.userPrefs`, and register that via our `PhoebusPreferenceService`.

 To keep using the basic API, it's fairly easy to implement an `InMemoryPreferences` class, register that in the `PhoebusPreferenceService`, so prefs are loaded from  `xxx_preferences.properties` files for build-in settings, then add values provided via `-settings /path/to/file`, and that's it. That "works", there are no surprises from settings saved by an older or even concurrently running instance. The `.memento` stuff still works to get the same window layout on restart.

There is one downside: In a few places we update preferences at runtime. It's mostly in the `DialogHelper` where some dialog sizes are persisted, so the next time you open that dialog, even after an app restart, it has the last used size.
My leaning is that's minor. The confusion I've had from saved preferences is much worse than dialogs coming back up with the default size.

PR #3041 (2024-06-17): In-memory preferences
Fixes #3040 

Removes the `FileSystemPreferences` that we copied from the Linux preference implementation with `InMemoryPreferences`.

Preferences are thus initialized from our `...preferences.properties` files, updated with optional `settings.ini` bundled in the product's "install" location, and finally updated via `-settings /path/to/xx.ini` command line options either in a launcher script or directly via the command line.

Preferences are no longer persisted, removing the surprises seen in the past when running a product with some `-settings ...` and then still seeing  [... cut]
closed 2024-06-17 by shroffk