ControlSystemStudio/phoebus #3058: phoebus.install variable in Settings.ini for Windows

XavSPM, 2024-06-24:
Hello,

I have a problem with the settings.ini file on Windows. To give you some context, I create installation packages with jpackage. Inside these packages, I include everything my programs need (interface, settings, icons, etc.).

On Linux, everything works correctly (in .deb and .rpm formats). However, on Windows, I encounter an error with my settings.ini file. Here is the content of the settings.ini file:

```
org.phoebus.ui/home_display=$(phoebus.install)\\ui\\main.bob?app=display_runtime
```

I use the `$(phoebus.install)` variable because it points to the installation location of my program. However, when I run the application, I get this error:

```
PS C:\Users\xgoiziou\phoebus_translate_fr\phoebus_product\target> java -jar .\produc-spr-4.7.4-SNAPSHOT.jar
2024-06-24 13:01:56 CONFIG [org.phoebus.product.Launcher] Loading settings from C:\Users\xgoiziou\phoebus_translate_fr\phoebus_product\target\settings.ini
2024-06-24 13:01:56 INFO [org.phoebus.product.Launcher] Phoebus (PID 24584)
2024-06-24 13:01:56 WARNING [javafx] Unsupported JavaFX configuration: classes were loaded from 'unnamed module @b92a00e'
2024-06-24 13:01:57 INFO [org.phoebus.channelfinder.ChannelFinderClient] Creating a channelfinder client to : http://localhost:8080/ChannelFinder
2024-06-24 13:01:57 INFO [org.phoebus.applications.display.navigation.DisplayNavigationViewApp] Loading Display Navigation View
2024-06-24 13:01:57 WARNING [org.phoebus.ui.application.PhoebusApplication] Cannot parse top resource 'C:\Users\xgoiziou\phoebus_translate_fr\phoebus_product\target\ui\main.bob?app=display_runtime'
java.net.URISyntaxException: Illegal character in opaque part at index 2: C:\Users\xgoiziou\phoebus_translate_fr\phoebus_product\target\ui\main.bob?app=display_runtime
        at java.base/java.net.URI$Parser.fail(URI.java:2995)
        at java.base/java.net.URI$Parser.checkChars(URI.java:3166)
        at java.base/java.net.URI$Parser.parse(URI.java:3202)
        at java.base/java.net.URI.<init>(URI.java:645)
        at org.phoebus.ui.application.TopResources.parse(TopResources.java:56)
        at org.phoebus.ui.application.PhoebusApplication.createToolbar(PhoebusApplication.java:944)
        at org.phoebus.ui.application.PhoebusApplication.startUI(PhoebusApplication.java:478)
        at org.phoebus.ui.application.PhoebusApplication.lambda$backgroundStartup$9(PhoebusApplication.java:463)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:456)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:455)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(WinApplication.java:185)
        at java.base/java.lang.Thread.run(Thread.java:1583)
```

And the home button does nothing.

I have tried several combinations, and the only one that worked is this: `org.phoebus.ui/home_display=/Users/xgoiziou/phoebus_translate_fr/phoebus_product/target/ui/main.bob?app=display_runtime`

However, this is not what I want because I lose the flexibility of using the `phoebus.install` variable.

Thank you for your help.

XavSPM, 2024-06-25:
I have continued my investigations and replaced the variable `$(phoebus.install)` with its value in the `settings.ini` file:

```
org.phoebus.ui/home_display=C:\Users\xgoiziou\Desktop\produc-spr-4.7.4-SNAPSHOT\ui\main.bob?app=display_runtime
```

However, I still get the same error. I then replaced the `\` with `\\`:

```
org.phoebus.ui/home_display=C:\\Users\\xgoiziou\\Desktop\\produc-spr-4.7.4-SNAPSHOT\\ui\\main.bob?app=display_runtime
```

But this did not solve the problem. I also tried:

```
org.phoebus.ui/home_display=C\:\\Users\\xgoiziou\\Desktop\\produc-spr-4.7.4-SNAPSHOT\\ui\\main.bob?app=display_runtime
```

Without success. The only solution that worked is this:

```
org.phoebus.ui/home_display=C:/Users/xgoiziou/Desktop/produc-spr-4.7.4-SNAPSHOT/ui/main.bob?app=display_runtime
```

While researching online, I found this discussion: [Java URI SyntaxException on Stack Overflow](https://stackoverflow.com/questions/2856126/java-net-urisyntaxexception). It seems there is an issue with interpreting paths on Windows.

I see two possibilities to solve this problem:
1. Modify the function that uses `java.net.URI`.
2. Change how the variable paths are recorded on Windows. This second solution would allow us to keep the same `settings.ini` files on both Linux and Windows.

What do you think?

kasemir, 2024-06-25:
There are a few sections where on one hand we aim to use a "URL", on the other hand we're trying to be flexible and allow  For example related displays, where we want to allow spaces in the relative path to another display, which then need to be patched as '%20' or the like.

The most practical solution here might be to add a replacement of '\' into '/' in a suitable location,

XavSPM, 2024-06-25:
I have noticed another bug. For directories containing spaces, they need to be replaced with %20.

Example:

`org.phoebus.ui/home_display=C:/Program Files/EPICS Arronax/app/ui/main.bob?app=display_runtime` should become `org.phoebus.ui/home_display=C:/Program%20Files/EPICS%20Arronax/app/ui/main.bob?app=display_runtime`.

@kasemir 
If you wish, I can take care of this modification. However, since Java is not my preferred language, I would need to know in which file to make these changes.

I think we could use a formatting function if the `java.vm.vendor` variable is equal to `Microsoft`.

kasemir, 2024-08-13:
Fine with me

closed 2024-09-03 by XavSPM