ControlSystemStudio/phoebus #3059: No JFX for linux-aarch64

kasemir, 2024-06-25:
Maven central doesn't appear to include JFX version 21.0.3 variants for linux-aarch64.

This breaks the EPICS training VM setup on Mac OS, https://github.com/epics-training/training-vm/blob/main/doc/creating-vm-from-scratch-apple-silicon.md :

```
Could not resolve dependencies..
The following artifacts could not be resolved:
org.openjfx:javafx-controls:jar:linux-aarch64:21.0.3,
org.openjfx:javafx-graphics:jar:linux-aarch64:21.0.3,
...
```

https://stackoverflow.com/questions/78263233/javafx-21-for-linux-aarch64 suggests that linux-aarch64 is available for some JFX updates but not others.

shroffk, 2024-08-26:
We have the option to revert to 21.0.1 which is supported or move forward to 22.0.2

XavSPM, 2024-09-08:
I encountered the same issue on a Raspberry Pi.

I was able to compile using versions 21.0.1 and 22.0.2.
However, since version 22 of JavaFX is not an [LTS](https://gluonhq.com/products/javafx/) version.

A choice must therefore be made.

ralphlange, 2026-11-14:
Phoebus v6.0.0 still pins openjfx.version 21.0.7, and 21.0.7 still has no linux-aarch64 jars.
21.0.11 does have linux-aarch64 jars, master has moved to 25.0.4.

Should the option of pinning to 21.0.11 on 6.0.0 for linux-aarch64 be documented somewhere?