ControlSystemStudio/phoebus #3069: Expose JFXRepresentation to other packages?

daykin, 2024-07-08:
For the 'click heat-mapping' aspect of the UX analytics plugin, I need to know the Zoom level and boundaries of the `JFXRepresentation` of a display runtime in order to reconcile click events between clients that may have different sized and zoomed instances. The most trivial solution would be to change `DisplayRuntimeInstance.getRepresentation()` from package-private to public. 

Alternatively, it might be safer from a compartmentalization perspective to only add a `getZoom()`, `getCurrentHeight()`, etc, but that is less flexible.

Which one would all of you Phoebus gurus suggest?

kasemir, 2024-07-11:
Make the DisplayRuntimeInstance.getRepresentation() public, maybe with a comment that direct access to the JFX representation was allowed for special use cases like UX analytics. Display code should try to keep the separation of Model, Runtime and Representation which, in the original implementation, allowed swapping between an SWT and JFX representation.

daykin, 2024-07-15:
#3077 opened