ControlSystemStudio/phoebus #3074: [bug] can not load IA32-bit .dll on a AMD 64-bit platform

konosubakonoakua, 2024-07-15:
version: phoebus 4.7.3
error: 
```cmd
2024-07-15 10:07:53 INFO [org.phoebus.product.Launcher] Phoebus (PID 29068)
2024-07-15 10:07:53 WARNING [javafx] Unsupported JavaFX configuration: classes were loaded from 'unnamed module @4bfdfda7'
Loading library api-ms-win-core-console-l1-1-0 from resource failed: java.lang.UnsatisfiedLinkError: C:\Users\...\.openjfx\cache\19+11\amd64\api-ms-win-core-console-l1-1-0.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
java.lang.UnsatisfiedLinkError: C:\Users\...\.openjfx\cache\19+11\amd64\api-ms-win-core-console-l1-1-0.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
```

georgweiss, 2024-07-22:
This looks very odd.
How did you build the Phoebus application?

konosubakonoakua, 2024-07-22:
I just download the release pkg here: [v4.7.3](https://github.com/ControlSystemStudio/phoebus/releases/tag/v4.7.3).

konosubakonoakua, 2024-07-22:
BTW, 4.7.2 works well

georgweiss, 2024-07-22:
OK, I can reproduce. The package was not correctly built for non-AMD platform. I will try to find time to upload a new package.

georgweiss, 2024-07-23:
Can you please try product-4.7.3.zip in https://github.com/ControlSystemStudio/phoebus/releases/edit/v4.7.3?

konosubakonoakua, 2024-08-08:
> Can you please try product-4.7.3.zip in https://github.com/ControlSystemStudio/phoebus/releases/edit/v4.7.3?

it works, ty.

ralphlange, 2025-01-18:
I'm getting the same error using the currently available download of 4.7.3. (18/Jan/2025)
```
C:\Users\langer\phoebus-4.7.3>echo off
2025-01-18 16:26:45 INFO [org.phoebus.product.Launcher] Phoebus (PID 27288)
2025-01-18 16:26:46 WARNING [javafx] Unsupported JavaFX configuration: classes were loaded from 'unnamed module @6d1cd7cc'
Loading library api-ms-win-core-console-l1-1-0 from resource failed: java.lang.UnsatisfiedLinkError: C:\Users\langer\.openjfx\cache\19+11\amd64\api-ms-win-core-console-l1-1-0.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
java.lang.UnsatisfiedLinkError: C:\Users\langer\.openjfx\cache\19+11\amd64\api-ms-win-core-console-l1-1-0.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
        at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
        at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:331)
        at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:197)
 ...
Is there anything special that I need to do?

ralphlange, 2025-01-18:
[image]

Works like a charm. Resolving.

closed 2025-01-18 by ralphlange

shroffk, 2025-01-18:
I rebuilt the win binaries. This was to verify that the correct javafx binaries are packaged with the product.

@ralphlange tested this to verify that the above problem is addressed.

@georgweiss I removed the product you had uploaded... We can reupload it if necessary

ralphlange, 2025-01-18:
Is that two different ways of building that are working for disjoint sets of users?
But my error messages were exactly the same as the OP's.