ControlSystemStudio/phoebus #3081: Multiple configs(topics) in alarm logger

jykim-physics, 2024-07-17:
Hello all,

I'm looking for how we can include 3 topics in one alarm logger.
As long as I know, 1 alarm server per topic is correct.
In case of alarm logger, it looks like we can notify multiple topics(`-topics`)

-topics topic1,topic2,topic3
I found that the third topic(topic3) do not send logger messages if there was not any error.
I could double check in case of -topics topic3,topic1,topic2, topic2 do not send messages.

It's a bit weird that the only third(last) topic didn't work properly. 
Could someone give comments on it? or I'm wondering if it's a known issue. Thanks!

jykim-physics, 2024-07-17:
I used 4.7.2 for all alarm servers and alarm loggers, but 4.7.4(the most recent version) looks no problem. 
May I ask if there were significant updates? Thanks a lot!

shroffk, 2024-07-19:
That is strange indeed...can you try the latest master and see if the problem persists

jykim-physics, 2024-07-24:
Thanks for your answer!
Sorry for stupid question. Where can I find those kinds of log messages? When I executed the logger with java command, there were no corresponding contents. 
And also I tried with the latest version, and it remained the same situation.

jykim-physics, 2024-08-08:
I found a solution! `thread_pool_size=6` option was needed since I need 3 topics. The default setting was 4.

closed 2024-08-08 by jykim-physics