ControlSystemStudio/phoebus #3085: Phoebus fails to connect to Save and Restore service

kristjansoln, 2024-07-22:
We are currently running version 4.7.2 of Phoebus along with the Save and Restore service in Docker containers. After upgrading both to 4.7.3, Phoebus fails to connect to the Save and Restore service, returning the 404 status error shown below. 

```
2024-07-22 14:04:33 SEVERE [org.phoebus.applications.saveandrestore.ui.SaveAndRestoreService] Unable to retrieve root node, cause: org.phoebus.applications.saveandrestore.SaveAndRestoreClientException: Failed : HTTP error code : 404, error message: <!doctype html><html lang="en"><head><title>HTTP Status 404 – Not Found</title><style type="text/css">body {font-family:Tahoma,Arial,sans-serif;} h1, h2, h3, b {color:white;background-color:#525D76;} h1 {font-size:22px;} h2 {font-size:16px;} h3 {font-size:14px;} p {font-size:12px;} a {color:black;} .line {height:1px;background-color:#525D76;border:none;}</style></head><body><h1>HTTP Status 404 – Not Found</h1></body></html>
```

Below is the configuration for the Save and Restore service used when starting Phoebus. Using the new setting variable (as described in #2838) did not fix the problem.

```bash
# Save and restore settings
# org.phoebus.applications.saveandrestore/jmasar.service.url=http://localhost:8080
org.phoebus.applications.saveandrestore.client/jmasar.service.url=http://localhost:8080
```

Save and Restore service is configured to listen on port 8080, and Elastic search is configured to listen on port 19200. Everything is running on the same machine and accessible via `localhost`. http://localhost:8080/save-restore/node/44bef5de-e8e6-4014-af37-b8f6c8a939a2 and http://localhost:8080/save-restore/ give the following responses:
```
{
  "uniqueId": "44bef5de-e8e6-4014-af37-b8f6c8a939a2",
  "name": "Root folder",
  "description": null,
  "created": "2024-07-22T11:47:41.279+00:00",
  "lastModified": "2024-07-22T11:47:41.279+00:00",
  "nodeType": "FOLDER",
  "userName": "anonymous",
  "tags": null,
  "properties": null
}
```
and
```
{
  "name": "@project.artifactId@",
  "version": "@project.version@",
  "elastic": {
    "status": "Connected",
    "clusterName": "epics-services",
    "clusterUuid": "X4H7xGjBQVu89nSkT-wV9Q",
    "version": "co.elastic.clients.elasticsearch._types.ElasticsearchVersionInfo@6a3d4f00"
  }
}
```

When inspecting sockets using `ss`, you can see that Save and Restore service has established connection to the Elastic Search service, and that Phoebus establishes the connection to the service when the Save and Restore tab is opened.
The PID of Save and Restore service is `3862425`.
The command `ss -tupna | grep 19200` returns the following
```
...
tcp   LISTEN     0      4096                [::ffff:127.0.0.1]:19200                   *:*     users:(("java",pid=3856040,fd=377))
tcp   LISTEN     0      4096               [::ffff:10.71.3.39]:19200                   *:*     users:(("java",pid=3856040,fd=376))
...
tcp   ESTAB      0      0                   [::ffff:127.0.0.1]:48726  [::ffff:127.0.0.1]:19200 users:(("java",pid=3862425,fd=27))
...
```
After opening the Save and Restore utility in Phoebus, the command `ss -tupna | grep 8080` returns
```
tcp   LISTEN     0      100                                  *:8080                    *:*     users:(("java",pid=3862425,fd=46))       
tcp   ESTAB      0      0                   [::ffff:127.0.0.1]:8080   [::ffff:127.0.0.1]:51546 users:(("java",pid=3862425,fd=50))       
tcp   ESTAB      0      0                   [::ffff:127.0.0.1]:51546  [::ffff:127.0.0.1]:8080  users:(("java",pid=3894982,fd=302))
```
with the PID `3894982` belonging to the Phoebus instance.

Here are the logs from the Save and Restore service, which also contain a Log4j error not finding a logging implementation.

```
Jul 22 13:47:39 services-VM docker[3856281]:  Container service-save-and-restore  Created
Jul 22 13:47:39 services-VM docker[3856281]: Attaching to service-save-and-restore
Jul 22 13:47:39 services-VM docker[3856281]: service-save-and-restore  | ERROR StatusLogger Log4j2 could not find a logging implementation. Please add log4j-core to the classpath. Using SimpleLogger to log to the console...
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |   .   ____          _            __ _ _
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |  /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  | ( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |  \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |   '  |____| .__|_| |_|_| |_\__, | / / / /
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |  =========|_|==============|___/=/_/_/_/
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |  :: Spring Boot ::                (v2.7.3)
Jul 22 13:47:40 services-VM docker[3856281]: service-save-and-restore  |
Jul 22 13:47:41 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:41.745  INFO 7 --- [           main] o.apache.catalina.core.StandardService   : Starting service [Tomcat]
Jul 22 13:47:41 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:41.747  INFO 7 --- [           main] org.apache.catalina.core.StandardEngine  : Starting Servlet engine: [Apache Tomcat/9.0.65]
Jul 22 13:47:41 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:41.819  INFO 7 --- [           main] o.a.c.c.C.[.[localhost].[/save-restore]  : Initializing Spring embedded WebApplicationContext
Jul 22 13:47:42 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:42.465  INFO 7 --- [           main] o.p.s.s.p.config.ElasticConfig           : Created index: saveandrestore_tree : acknowledged true
Jul 22 13:47:42 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:42.590  INFO 7 --- [           main] o.p.s.s.p.config.ElasticConfig           : Created index: saveandrestore_configuration : acknowledged true
Jul 22 13:47:42 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:42.711  INFO 7 --- [           main] o.p.s.s.p.config.ElasticConfig           : Created index: saveandrestore_snapshot : acknowledged true
Jul 22 13:47:42 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:42.833  INFO 7 --- [           main] o.p.s.s.p.config.ElasticConfig           : Created index: saveandrestore_composite_snapshot : acknowledged true
Jul 22 13:47:42 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:42.957  INFO 7 --- [           main] o.p.s.s.p.config.ElasticConfig           : Created index: saveandrestore_filter : acknowledged true
Jul 22 13:47:43 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 18:47:43.075  INFO 7 --- [           main] o.p.s.s.p.config.ElasticConfig           : Root node created
Jul 22 14:11:43 services-VM docker[3856281]: service-save-and-restore  | 2024-07-22 19:11:43.560  INFO 7 --- [nio-8080-exec-3] o.a.c.c.C.[.[localhost].[/save-restore]  : Initializing Spring DispatcherServlet 'dispatcherServlet'
```

Any help on the matter is much appreciated.

GDH-ISIS, 2024-07-22:
Am not certain on this, but had a similar experience. It was related to a change in the save and restore url.
I would recomment trying the following in your phoebus settings file : 
org.phoebus.applications.saveandrestore.client/jmasar.service.url=http://localhost:8080/save-restore

shroffk, 2024-07-22:
> Am not certain on this, but had a similar experience. It was related to a change in the save and restore url. I would recomment trying the following in your phoebus settings file : org.phoebus.applications.saveandrestore.client/jmasar.service.url=http://localhost:8080/save-restore

@kristjansoln note the extra `save-restore` in the URL

kristjansoln, 2024-07-23:
This fixed the problem. Thank you very much.

closed 2024-07-23 by kristjansoln