ControlSystemStudio/phoebus #3086: Data Browser : Max and min data values always applied to y scale when time axis changed

GDH-ISIS, 2024-07-23:
With the Phoebus Data Browser, when the Y axis autoscale is not selected, changing the x axis timescale appears to always autoscale the y value axis to the max/min of the data, not maintaining the max and min values previously defined. I believe this might be a bug (using a recent version of Phoebus compiled from source, running on Windows).

kasemir, 2024-07-23:
Under Properties, Value Axes, check if you have selected "When archived data arrives: ( ) Do nothing  (*) Stagger".

When you change the time axis, we fetch archived data for that new time range.  "(*) Stagger" then performs a Y axis update so you can see the stuff you just fetched from the archive, but then people will complain that the Y axis changed, see above. If you select "(*) Do nothing", other or even the same people will complain that they can't see the data. So you try different default values for this setting (see https://control-system-studio.readthedocs.io/en/latest/preference_properties.html#trends-databrowser3)
to minimize the number of emails that you get regarding this issue:

```
# Re-scale behavior when archived data arrives: NONE, STAGGER
org.csstudio.trends.databrowser3/archive_rescale=STAGGER
```

GDH-ISIS, 2024-07-23:
This looks perfect. I agree with the debate about what people want. Never ending cycle. Much appreciated.

closed 2024-07-23 by GDH-ISIS