ControlSystemStudio/phoebus #3099: No user message if settings are not found

georgweiss, 2024-07-31:
With the recent changes to properties/settings handling, user might end up in a situation where startup of Phoebus fails without any useful error message. For instance:

- Phoebus is installed from a package prepared by ```jpackage```.
- Package contains ```settings.ini``` in install directory.
- Launcher includes ```-settings /user/home/my-settings.ini```
- Launcher does not add console.
- If  ```/user/home/my-settings.ini``` does not exists, Phoebus will shut down and no message is available as console is not included in the launcher configuration.

In this case user _must_ be aware of the requirement to create a file (even if empty) ```/user/home/my-settings.ini```.

It would be nice to be able to show an ```Alert``` at the point where the Phoebus UI launches.

kasemir, 2024-07-31:
A swing dialog from the phoebus launcher for `-settings /bad/path/missing_file.ini` would be OK.
But that's where I'd stop, and not try to add more intelligence to this.

You can already provide a launcher for your site. In there, you decide which mandatory and optional settings you want to support.
If you want to support $HOME/my-settings.ini, you need to handle that in your launcher.
Somewhat like this if they should be optional:
```
#!/bin/bash

OPT=""
if [ -r "$HOME/my-settings.ini" ]
then
  OPT+="-settings $HOME/my-settings.ini"
fi
...
java -jar ... $OPT "$@"
```

.. or like this if it should be required:

```
...
else
 xmessage "Cannot find  $HOME/my-settings.ini"
 exit 1
fi
```

Trying to handle all this in the phoebus launcher would mean that we need `-optional_settings /path/to/something.ini` in addition to `-settings`, then maybe options to decide how an error should be displayed, all of which is site specific and can already be handled in the script that starts phoebus.

georgweiss, 2024-07-31:
The point is to avoid console and start script, and ```xmessage``` would be Linux specific (?). 
My Mac and Windows users just install Phoebus from IT management systems, and then (rightly) expect it to launch. So I need a way to tell them that they did not read the instructions of putting a settings file where it is supposed to be.

shroffk, 2024-07-31:
> A swing dialog from the phoebus launcher for -settings /bad/path/missing_file.ini would be OK.
But that's where I'd stop, and not try to add more intelligence to this.

+1

I think a swing dialog would fix your use case right... we could also create a crash log ~/.phoebus/logs/why_phoebus_did_not run
and the dialog can point to it

I have encountered a variety of problems of this nature where people try to install our Phoebus on their own. It would be a lot of work to figure out all of them ( missing file, file present but permissions are wrong, etc... add remote file systems and other infrastructure specific things can result in a lot of unique issues )

georgweiss, 2024-07-31:
OK, thanks for the feed-back.

I will implement this based on this discussion.