ControlSystemStudio/phoebus #3107: Document how to change port for alarm logger

minijackson, 2024-08-07:
Not long ago, I fell into a rabbit hole on why my `server.port` property was not recognized by Phoebus alarm logger.

After a while, I found out that it's because the `-properties /path/to/config.properties` argument doesn't forward the loaded properties to SpringBoot. So my solution was to also specify the environment variable `SPRING_CONFIG_IMPORT=/path/to/config.properties`.

It would be nice if there was an official, documented way of doing this, as port 8080 is used by quite a number of EPICS-related services by default.

minijackson, 2024-08-07:
I found https://github.com/ControlSystemStudio/phoebus/issues/2277#issuecomment-1137429703 recently, but as far as I know the `-Dserver.port` argument is not documented on the Phoebus documentation, and it might get quite annoying if one wants to modify other SpringBoot settings.

kasemir, 2024-08-07:
Yes, SpringBoot adds too many dependencies and obfuscates the configuration. We should remove it and have all services use a plain "main()" like the alarm server, scan server, RDB archive engine

shroffk, 2024-08-07:
Umm... There are quite a few other benefits of springboot, we are trying to merge 2 startup and configuration processes here which has resulted in some issues. I will look into cleaning it up.

kasemir, 2024-08-07:
Nah, it's a complex framework just like Eclipse was. It expects you to know which annotations to add and where to place files without a way to follow breadcrumbs or call trees. In ~4 years there'll be a springboot update that forces you re-do everything.

georgweiss, 2024-08-07:
We should make all service Spring Boot-based.

jeonghanlee, 2025-03-04:
So,we all have the 8080 ports for all services soon, right? @georgweiss ;)