ControlSystemStudio/phoebus #3108: backward compatibility with macros setting via scripts

shroffk, 2024-08-07:
The old `DataUtil.createMacrosInput` API used to provide a map which is not supported in the `MacrosWidgetProperty.setValueFromObject`

solution 1. change the DataUtil to return a macro object instead of a map  
```
    /**Create a MacrosInput, which can be used as the macros input for a container widget or display.
     * New macro can be added or replaced by
     * <code>MacrosInput.put(String macroName, String macroValue);</code>
     * @param include_parent_macros If parent macros should be included.
     * @return a new created {@link Macros}.
     */
    public final static Macros createMacrosInput(boolean include_parent_macros)
    {
        return new Macros();
    }
}
```
but this would require **all** scripts to be updates to replace the .put() with .add()

solution 2. add support to accept `if (value instanceOf Map)`  
```
    /** @param value Must be {@link org.csstudio.display.builder.model.spi.ActionInfo} array(!), not List */
    @Override
    public void setValueFromObject(final Object value) throws Exception
    {
        if (value instanceof Macros)
            setValue((Macros) value);
        else
            throw new Exception("Need Macros, got " + value);
    }
```

What is the reason for not supporting List/Maps here, this solution would not require existing scripts to be updated

kasemir, 2024-08-07:
Macros can have order.
`A=1,B=2,T=$(A),A=$(B),B=$(T)` and you end up with 2 for `$(A)` and ` for `$(B)`.
In a map, you'd only have one setting per key.

So a Map cannot hold the full detail that's possible with `Macros`.
But we should still allow `setValueFromObject` to read a `Map` in addition to `Macros`.

I'm even OK with `setValueFromObject` allowing a `String` of the form `name=value,name=value,...` because that would be the easiest way to get one or two macros defined from within a script.

shroffk, 2024-08-07:
Ok, 
Thank you, this makes a lot of sense.

I will add the support for the map and the string to the setValueFromObject

closed 2024-08-08 by shroffk