ControlSystemStudio/phoebus #3127: Yet another way to specify settings file?

georgweiss, 2024-09-03:
With the recent changes where ability to import settings has been removed, it is not possible to build an installer with ```jpackage``` using the ```$HOME``` symbol to specify a location of a settings file. At least not on Mac OS as ```$HOME``` is not expanded but rather used as-is.

At the same time, instructing users to maintain their settings in a file like:
```/Applications/CSS-Phoebus.app/Contents/app/settings.ini``` feels a bit awkward.

Suggesting to update the implementation of ```-settings /path/to/foo.ini``` to look for ```foo.ini``` in user's home directory if it turns out that ```/path/to/foo.ini``` is incorrect.

kasemir, 2024-09-03:
When a complete path is given as in `-settings /path/to/foo.ini`, we should not second-guess. An absolute path is pretty well defined.

But what about looking in the home directory for the case of `-settings foo.ini`?
Or in the "current" directory and the "home" folder? Might want to define the order: First current, then home?

georgweiss, 2024-09-03:
> But what about looking in the home directory for the case of ```-settings foo.ini```?

Actually that was my intention, just phrased it a bit wrong. Order "current" before "home" is fine.

georgweiss, 2024-09-03:
#3128

closed 2024-09-03 by georgweiss