ControlSystemStudio/phoebus #3142: Unable to locate authorization.conf file from Phoebus

GDH-ISIS, 2024-09-20:
Our latest build of a Phoebus and the creation of an installer using jpackage is unable to reach the authorization.conf file. It is thus unable to lauch the screen editor and access color and class definition files.

Details are :

- Installed on Windows operating system
- Latest Phoebus built from git clone from github
- Installer created using jpackage - installer copies settings.ini and authorization.conf file into install location
- setting.ini use $(phoebus.install)

Would it be possible to confirm the issue please? We have been stable to quite a long time and our settings have not been changed.

GDH-ISIS, 2024-09-20:
Old version that worked reported from about/preference settings/ window within Phoebus application
org.phoebus.security/authorization_file=$(phoebus.install)/ISIS/authorization.conf
Is there a change I am not aware of?

shroffk, 2024-09-20:
I think this might be related to
https://github.com/ControlSystemStudio/phoebus/issues/3058

GDH-ISIS, 2024-09-20:
I agree, could not write this out quickly enough :)

shroffk, 2024-09-20:
I tried the following

```
# Display Builder
org.phoebus.ui/home_display=$(phoebus.install)/../resources/Display_Builder/01_main.bob
```

and it converted it properly to

`org.phoebus.ui/home_display=C:/git/cs-studio/phoebus/../resources/Display_Builder/01_main.bob`

GDH-ISIS, 2024-09-20:
Would it be possible to try a location that contains a space please?

shroffk, 2024-09-20:
```
# Display Builder
org.phoebus.ui/home_display=$(phoebus.install)/../resources/Display_Builder/with space/01_main.bob
```

translated to

```
org.phoebus.ui/home_display=C:/git/cs-studio/phoebus/../resources/Display_Builder/with%20space/01_main.bob
```

This is working for me.
What is the raw preference you are using?

GDH-ISIS, 2024-09-20:
I think this is what you are looking for.

org.phoebus.security/authorization_file=$(phoebus.install)/ISIS/authorization.conf

shroffk, 2024-09-20:
Hmm...The resolved url looks correct to me.

> org.phoebus.security/authorization_file=C:/Program%20Files/Phoebus-ISIS/app/ISIS/authorization.conf

If you hard code this location does it work?

GDH-ISIS, 2024-09-20:
This was my interpretation. Let me test your request ...

GDH-ISIS, 2024-09-20:
The full path entered in settings.ini appears to work. i.e.
org.phoebus.security/authorization_file=C:/Program Files/Phoebus-ISIS/app/ISIS/authorization.conf

GDH-ISIS, 2024-09-20:
OK, let me confirm and make sure I have not done something silly. I will come back to you. Thanks. G

GDH-ISIS, 2024-09-20:
After clarification, with the latest build/compiled/jpackage/installed version into Program Files (java.version = 21.0.3):
**Working**
**settings.ini**                  org.phoebus.security/authorization_file=C:/Program Files/Phoebus-ISIS/app/ISIS/authorization.conf
**Phoebus (about dialog)**      org.phoebus.security/authorization_file=C:/Program Files/Phoebus-ISIS/app/ISIS/authorization.conf
**phoebus.install (about dialog)**  phoebus.install = C:\Program Files\Phoebus-ISIS\app

**Not working**
**settings.ini**                  org.phoebus.security/authorization_file=$(phoebus.install)/ISIS/authorization.conf
**Phoebus (about dialog)**      org.phoebus.security/authorization_file=C:/Program%20Files/Phoebus-ISIS/app/ISIS/authorization.conf
**phoebus.install (about dialog)**  phoebus.install = C:\Program Files\Phoebus-ISIS\app

We do appear to be suffering from the space (%20) in Program Files - at least from what I can see. My apologies if this is misleading. I may hard code the paths for the moment.

shroffk, 2024-09-20:
I think I have found the issue.
I believe the AuthorizationService which reads the file cannot process the paths with the special chars parsed (%20). I do not believe anything in $(...) is the cause of the issue.

We might want to look into if the AuthorizationServices should use an filename( with spaces and everything) or use valid URI's (file:///C:/Program%20Files/...)

GDH-ISIS, 2024-09-20:
This is good to know. Your suggestion looks to fall in line with other services and would make it common. Thumbs up to the suggestion.

GDH-ISIS, 2024-09-20:
I can resolve my immediate problem with a minor modification before compiling, although I have yet to decide whether I will use it.

sed -i 's|.replace(" ", "%20");|;|' /phoebus/core/framework/src/main/java/org/phoebus/framework/preferences/PropertyPreferenceLoader.java

shroffk, 2024-09-24:
It should be noted that other apps like the Display Builder Runtime are handling this change well. We might want to use the same resource loading

katysaintin, 2025-02-03:
> The full path entered in settings.ini appears to work. i.e. Am now beginning to doubt myself ... org.phoebus.security/authorization_file=C:/Program Files/Phoebus-ISIS/app/ISIS/authorization.conf

On Windows, you have to doble "/" in the path to make it works , in your case it's become : 
C://Program Files//Phoebus-ISIS//app//ISIS//authorization.conf