ControlSystemStudio/phoebus #3143: rename 'save snapshot' in context menu

kathryn-baker, 2024-09-24:
Following the discussion around save-and-restore at the EPICS Collaboration meeting, would it be possible to rename the 'Save Snapshot' option in the context menu to something like 'Save **Screenshot**'? I think this might help reduce confusion for anyone new to Phoebus and save-and-restore as this 'snapshot' is taking a snapshot of the screen rather than a snapshot of the PV values. 

[image]

georgweiss, 2024-09-24:
Agree completely

kasemir, 2024-09-24:
Yes. Actually, "Save Screenshot ...", ending with "..." to indicate that it's not click&done. You'll be prompted for a file name etc.

Just like "Save" (no "...") simply saves, while "Save As..." (with "...") will take a little longer.
Similarly, "Create Log..." and "Send Email ..." need "..." because they have some follow-up interaction.

kasemir, 2024-09-24:
Thank you for volunteering to clean that up!