ControlSystemStudio/phoebus #3150: There was an error while compiling the app/display/model in Maven.

SunamaZero, 2024-09-28:
[image]

[ERROR]   AdvancedConverterTest.listScriptFilesTest:37 expected: <true> but was: <false>

It seems that there is an error in the "listScriptFilesTest" method of the "AdvancedConverterTest" class regarding the assertion "assertTrue(listScriptFiles.size() == 4)".

SunamaZero, 2024-09-28:
[ERROR]   AdvancedConverterTest.listScriptFilesTest:37 expected: <true> but was: <false>

SunamaZero, 2024-09-28:
I compiled this condition to assertTrue(listOpiFiles.size() <= 4)

georgweiss, 2024-09-30:
I think commit d3d155d4486921618505bb18fdaadbdc991d9e62 is erroneous: constants for extensions were modified for OPI files, but not for script files. So listing of script files will fail as the code checks for ```endsWith("." + ext)``` (line 136).
So either the constants need to be updated, or the code should read ```endsWith(ext)```

katysaintin, 2024-09-30:
@georgweiss @SunamaZero , Unit Test is fixed. The extension file name should not include the dot caracter. Sorry for this unintentional oversight

georgweiss, 2024-09-30:
@katysaintin, can you please make that a PR to this repo?

SunamaZero, 2024-10-01:
@georgweiss @katysaintin I can't thank you enough. This issue has troubled me for a long time, and my English skills are limited. Words alone can't express how much I appreciate your help.

georgweiss, 2024-10-01:
You're welcome.

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

closed 2024-10-01 by shroffk

katysaintin, 2024-10-01:
> @georgweiss @katysaintin I can't thank you enough. This issue has troubled me for a long time, and my English skills are limited. Words alone can't express how much I appreciate your help.

@SunamaZero , you're welcome, sorry about that, we forget to test ours modifications . And thanks to our Unit Test, we found a bug in or Convertor. So thank you to you.