ControlSystemStudio/phoebus #3156: Drop-downs for menus or multi-entry action buttons are cut off on Linux

kasemir, 2024-10-04:
On Linux, drop-downs for menus or multi-entry action buttons are cut off.
You see about half of the bottom entry.

[image]

[image]

[image]

This happens with the currently used javafx 22.0.2. Did not happen with 21.0.3.

Looks like the wheels are falling off JavaFX running on Linux.
See  #3132 for drag/drop breaking the mouse.

kasemir, 2024-10-04:
Might be related to https://bugs.openjdk.org/browse/JDK-8329821, "menus are incorrectly sized", assumed fixed in JFX 23

kasemir, 2024-10-04:
Indeed, update to jfx 23 fixes the issue.

ButJavaFX 23 Requires JDK 21 or later and has some more changes in the release notes, https://github.com/openjdk/jfx/blob/jfx23/doc-files/release-notes-23.md

kasemir, 2024-10-07:
Going back to the older version 21.0 of the JFX graphics library also avoids this issue and doesn't require a different JDK.

shroffk, 2024-10-07:
For the time being I agree with going back to 21.0

Overall, I think this is a mixed bag of good and bad news... things are being fixed ( hopefully improved ) at a reasonable pace. However, it seems to push a upgrade model that might be a bit faster than we are used to or have planned for.

PR #3208 (2024-12-13): Downgrade to JavaFX 21.0.3
As discussed in monthly meerting 2024-12-11. Once 5.x of the project has been released to Maven, upgrade to JavaFX 23 (or 24?) and JDK 21+ is projected.

Resolves #3156
closed 2024-12-13 by shroffk