ControlSystemStudio/phoebus #3157: .git directory is missing

PeiyuQuan, 2024-10-04:
When i installed Phoebus, i got an error.

install-jars: .git directory is not found!, There is no .git directory in dependencies. How to fix this issue? Thank you.

(phoebus-community) [specadm@bluecameralx phoebus]$ mvn clean verify -f dependencies/pom.xml 
[INFO] Scanning for projects...
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] dependencies                                                       [pom]
[INFO] install-jars                                                       [jar]
[INFO] phoebus-target                                                     [jar]
[INFO] 
[INFO] ----------------------< org.phoebus:dependencies >----------------------
[INFO] Building dependencies 4.7.4-SNAPSHOT                               [1/3]
[INFO]   from pom.xml
[INFO] --------------------------------[ pom ]---------------------------------
[INFO] 
[INFO] --- clean:3.2.0:clean (default-clean) @ dependencies ---
[INFO] 
[INFO] --- git-commit-id:9.0.1:revision (get-the-git-infos) @ dependencies ---
[INFO] 
[INFO] --- failsafe:2.22.0:integration-test (default) @ dependencies ---
[INFO] Tests are skipped.
[INFO] 
[INFO] --- failsafe:2.22.0:verify (default) @ dependencies ---
[INFO] Tests are skipped.
[INFO] 
[INFO] ----------------------< org.phoebus:install-jars >----------------------
[INFO] Building install-jars 4.7.4-SNAPSHOT                               [2/3]
[INFO]   from install-jars/pom.xml
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- clean:3.2.0:clean (default-clean) @ install-jars ---
[INFO] 
[INFO] --- install:2.5.2:install-file (install-pbraw) @ install-jars ---
[INFO] Installing /home/specadm/phoebus/dependencies/install-jars/lib/pbraw/pbrawclient-0.0.10.jar to /home/specadm/.m2/repository/org/epics/pbrawclient/0.0.10/pbrawclient-0.0.10.jar
[INFO] Installing /home/specadm/phoebus/dependencies/install-jars/lib/pbraw/pom.xml to /home/specadm/.m2/repository/org/epics/pbrawclient/0.0.10/pbrawclient-0.0.10.pom
[INFO] 
[INFO] --- git-commit-id:9.0.1:revision (get-the-git-infos) @ install-jars ---
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for dependencies 4.7.4-SNAPSHOT:
[INFO] 
[INFO] dependencies ....................................... SUCCESS [  0.517 s]
[INFO] install-jars ....................................... FAILURE [  0.132 s]
[INFO] phoebus-target ..................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  0.765 s
[INFO] Finished at: 2024-10-04T14:20:29-07:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal io.github.git-commit-id:git-commit-id-maven-plugin:9.0.1:revision (get-the-git-infos) on project install-jars: .git directory is not found! Please specify a valid [dotGitDirectory] in your project -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :install-jars

PeiyuQuan, 2024-10-07:
Hellp Shroffk

What modification did you do? I got the same error.
Thx

[INFO] install-jars ....................................... FAILURE [  0.384 s]
[INFO] phoebus-target ..................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  2.700 s
[INFO] Finished at: 2024-10-07T13:44:03-07:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal io.github.git-commit-id:git-commit-id-maven-plugin:9.0.1:revision (get-the-git-infos) on project install-jars: .git directory is not found! Please specify a valid [dotGitDirectory] in your project -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :install-jars

PeiyuQuan, 2024-10-07:
Hello Kunal

I have remove system.out calls, but i got the same errors.

Peiyu

shroffk, 2024-10-08:
https://github.com/ControlSystemStudio/phoebus/pull/3158/files#diff-9c5fb3d1b7e3b0f54bc5c4182965c4fe1f9023d449017cece3005d3f90e8e4d8

Add the following property

```
 <configuration>
          <failOnNoGitDirectory>false</failOnNoGitDirectory>
          ...
```

PeiyuQuan, 2024-10-08:
It works. Thank you. Kunal

PR #3158 (2024-10-09): make git properties optional
fixes #3157