ControlSystemStudio/phoebus #3160: Documentation pointing to /latest/, but /stable/ documentation seems better

PelloUsabiaga, 2024-10-08:
Hi, I just want to point that the default link to the docs points to the /latest/ documentation, which, at least in my opinion, is much worse than the one in /stable/. I don't know if this is the expected behaviour, or if it has been discussed before, and if the /latest/ one is the one that wants to be displayed to new users like myself, but if it is not, consider changing it. 

Thank you!

shroffk, 2024-10-08:
Can you help us identify where the latest documentation diverges ( and worsens ) from the stable documentation.
We should be able to easily switch to a stable release, but I would like to keep the latest documentation as clean and useable as possible (since the latest will become the next stable with the upcoming release )

PelloUsabiaga, 2024-10-09:
I have been reviewing a little, and I am sorry, I have to retire what I said about that it is much worse. It has been mainly my problem, because I was not able to find the different pages of the documentation and their contents efficiently. The contents are, indeed, similar and I guess that more updated on the /latest/ version. But the index on the left lateral of the page does really help to navigate the docs and check its contains, without the need to constantly go back to the index page. I guess you have had your reasons to switch the sphinx theme, but at least in my opinion, the previous one had a better look and feel, and if you can introduce the lateral index back the new one would be much more navigable.

Thank you by the way for the amazing tool phoebus is! I have been testing it, and it works very smoothly!

shroffk, 2024-10-09:
> introduce the lateral index back the new one would be much more navigable

I think I agree with you, we should try to get the lateral index back

[image]

PR #3380 (2025-04-24): docs: switch to Read the Docs theme
Fixes #3160.

This switches the documentation from the "default" theme to the "Read the Docs" theme, which provides a sidebar that is consistent across all pages, which makes it easier to navigate, and easier to grasp the structure of the documentation.

There are other Sphinx themes, which you can see here: https://sphinx-themes.org/
but the RTD theme is very popular, and well designed, in my opinion.
closed 2025-04-24 by shroffk