ControlSystemStudio/phoebus #3192: Phoebus Alarm Table : Description query

GDH-ISIS, 2024-11-15:
May I please double check something please - the alarm table that we have configured looks to be stable and up to date. However, the Description column contains the PV name. May I just query whether I have something configured wrongly. The PV description string does not appear.

We have the following in our client settings  (Windows), which I believe is the default
org.phoebus.applications.alarm/alarm_table_columns=Icon, PV, Description, Alarm Severity, Alarm Status, Alarm Time, Alarm Value, PV Severity, PV Status

georgweiss, 2025-01-07:
The description is read from the alarm configuration file, or set to PV name if not present. It is *not* the DESC field of the PV record.

GDH-ISIS, 2025-01-13:
Ah, that would answer my query. Much appreciated.