ControlSystemStudio/phoebus #3198: Elog - Log Entry Calender and Table exceptions

sifriede, 2024-11-22:
Dear Phoebus team and developer,

## Short description
I get exceptions, when I try to use the Log Entry Calender and Table with elog.

## My environment
### Phoebus version
I used the latest release version:
https://github.com/ControlSystemStudio/phoebus/releases/download/v4.7.3/Phoebus-4.7.3-linux.tar.gz

### OS
```
OS=Debian 12 (bookworm)
JAVA_HOME=/usr/lib/jvm/java-17-openjdk-amd64
CS STUDIO Version=4.7.3
JAVA VERSION=Oracle Corporation 17.0.13+11-Debian-2deb12u1
JAVA FX Version = 19+11
```
### settings.ini
```
org.phoebus.logbook.ui/logbook_factory=elog
org.phoebus.logbook/logbook_factory=elog
org.phoebus.elog.api/elog_url=http://elog.mesa.kph/mesa/
org.phoebus.elog.api/categories=
org.phoebus.elog.api/types=
org.phoebus.elog.api/username=mesa
org.phoebus.elog.api/password=<actually_hashed_password_read_from_elog_passwordfile>
```

### elog logbook config
Based on Florian Feldbauers example
```
Attributes = Author, Type, Category, Subject
Options Type = Routine, Software Installation, Problem Fixed, Configuration, Other
Options Category = General, Hardware, Software, Network, Other, TODO, STEAM
Extendable Options = Category
Required Attributes = Author, Type
Page Title = ELOG - $subject
```

## Problem description
Using "Send to Logbook" I can successfully add entries to the elog. If I click on logbooks or tags I retrieve the corresponding types and categories defined in elog automatically. I see these entries in the elog web view.

But if I use the Log Entry Calender or Table, I retrieve the following exceptions and don't see any log entries:

### Log Entry Calender
```
org.phoebus.logbook.LogbookException: java.lang.UnsupportedOperationException
	at org.phoebus.logbook.LogClient.search(LogClient.java:494)
	at org.phoebus.logbook.olog.ui.LogbookSearchJob.lambda$getRunnable$0(LogbookSearchJob.java:63)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:840)
Caused by: java.lang.UnsupportedOperationException
```

### Log Entry Table
```
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException: Cannot invoke "java.lang.Exception.getMessage()" because "exception" is null
	at org.phoebus.ui.dialog.ExceptionDetailsErrorDialog.doOpenError(ExceptionDetailsErrorDialog.java:96)
	at org.phoebus.ui.dialog.ExceptionDetailsErrorDialog.lambda$openError$1(ExceptionDetailsErrorDialog.java:62)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:457)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:456)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:316)
	at java.base/java.lang.Thread.run(Thread.java:840)
```
	
Thanks in advance.
Best regards
Simon

georgweiss, 2024-11-22:
In both applications the UI invokes a search based on the default query, e.g. search for log entries for the latest 12 hours.
However, the elog ```LogClient``` does not implement the ```search``` method,  and the default implementation throws an ```UnsupportedException```.

I think you would need to reach out to @ffeldbauer who I think is the original contributor of the elog module.

kasemir, 2024-11-22:
Aside from the logbook error, the `ExceptionDetailsErrorDialog` should be fixed.
You ought to see the error dialog, in this case then showing that there's an `UnsupportedOperationException`.

Not sure how the `ExceptionDetailsErrorDialog` is called in this case, maybe the exception needs to be passed on.
As a fallback, `doOpenError` should check if the exception is null and not crash. 
https://github.com/ControlSystemStudio/phoebus/blob/master/core/ui/src/main/java/org/phoebus/ui/dialog/ExceptionDetailsErrorDialog.java

sifriede, 2024-11-22:
Dear George,

In a talk by Florian he showed,that it did work in 2020 with the search string.
https://indico.fhi-berlin.mpg.de/event/52/contributions/564/attachments/222/683/phoebus-elog-client.pdf

As far as I know, Florian is not actively maintaining the elog app anymore.

Do you have any idea what could have changed this behavior? Elog was fixed for newer versions of Phoebus in the codeathon 2023, right?

Best regards
________________________________
Von: Georg Weiss
Gesendet: Fri Nov 22 18:41:38 GMT+01:00 2024
An: ControlSystemStudio/phoebus
Cc: "Friederich, Dr. Simon" , Author
Betreff: Re: [ControlSystemStudio/phoebus] Elog - Log Entry Calender and Table exceptions (Issue #3198)

In both applications the UI invokes a search based on the default query, e.g. search for log entries for the latest 12 hours.
However, the elog LogClient does not implement the search method, and the default implementation throws an UnsupportedException.

I think you would need to reach out to @ffeldbauer<https://github.com/ffeldbauer> who I think is the original contributor of the elog module.

georgweiss, 2024-11-22:
OK.
Though I have not confirmed it, I think this could be a mismatch between the build and your settings. While the tar ball you reference includes the newer Olog UI, the elog module depends on the older UI as implemented in ```app-logbook-ui```.  In other words, you need to build a product that includes that module and excludes ```app-logbook-olog```

sifriede, 2024-11-22:
Dear Georg,

I have now downloaded the Source file and I have compiled 
phoebus-product as is:

...
     [INFO] app-logbook ........................................ SUCCESS 
[  0.004 s]
     [INFO] app-logbook-inmemory ............................... SUCCESS 
[  1.245 s]
     [INFO] app-logbook-ui ..................................... SUCCESS 
[  3.257 s]
     [INFO] app-logbook-olog ................................... SUCCESS 
[  0.003 s]
     [INFO] app-logbook-olog-client ............................ SUCCESS 
[  1.114 s]
     [INFO] app-logbook-olog-client-es ......................... SUCCESS 
[  1.661 s]
     [INFO] app-logbook-olog-ui ................................ SUCCESS 
[  4.162 s]
     [INFO] app-logbook-elog ................................... SUCCESS 
[  0.862 s]
...

I guess, I have now all apps included. Unfortunately, the error remains.

Best regards

--------------xaZjcIhMX4F2J6nFbnOrUo5K
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: 8bit

<!DOCTYPE html>
<html>
  <head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
  </head>
  <body>
    <p><font face="Helvetica, Arial, sans-serif">Dear Georg,</font></p>
    <p><font face="Helvetica, Arial, sans-serif">I have now downloaded
        the Source file and I have compiled phoebus-product as is:</font></p>
    <p><font face="Helvetica, Arial, sans-serif">...<br>
            [INFO] app-logbook ........................................
        SUCCESS [  0.004 s]<br>
            [INFO] app-logbook-inmemory ...............................
        SUCCESS [  1.245 s]<br>
            [INFO] app-logbook-ui .....................................
        SUCCESS [  3.257 s]<br>
            [INFO] app-logbook-olog ...................................
        SUCCESS [  0.003 s]<br>
            [INFO] app-logbook-olog-client ............................
        SUCCESS [  1.114 s]<br>
            [INFO] app-logbook-olog-client-es .........................
        SUCCESS [  1.661 s]<br>
            [INFO] app-logbook-olog-ui ................................
        SUCCESS [  4.162 s]<br>
            [INFO] app-logbook-elog ...................................
        SUCCESS [  0.862 s]<br>
        ...<br>
      </font></p>
    <p><font face="Helvetica, Arial, sans-serif">I guess, I have now all
        apps included. Unfortunately, the error remains.<br>
      </font></p>
    <p><font face="Helvetica, Arial, sans-serif">Best regards<br>
      </font></p>
    <div class="moz-cite-prefix">Am 22.11.2024 um 20:13 schrieb Georg
      Weiss:<br>
    </div>
    <blockquote type="cite"
***@***.***">
      <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
      <p dir="auto">OK.<br>
        Though I have not confirmed it, I think this could be a mismatch
        between the build and your settings. While the tar ball you
        reference includes the newer Olog UI, the elog module depends on
        the older UI as implemented in <code class="notranslate">app-logbook-ui</code>.
        In other words, you need to build a product that includes that
        module and excludes <code class="notranslate">app-logbook-olog</code></p>
      <p
style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
href="https://github.com/ControlSystemStudio/phoebus/issues/3198#issuecomment-2494607006"
          moz-do-not-send="true">view it on GitHub</a>, or <a
          moz-do-not-send="true">unsubscribe</a>.<br>
src="https://github.com/notifications/beacon/AHDLLDKSN5S53FRCJQ3OY332B565LA5CNFSM6AAAAABSJUOPBCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUUWCXJ4.gif"
          height="1" width="1" alt="" moz-do-not-send="true"><span
style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;ControlSystemStudio/phoebus/issues/3198/2494607006</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/ControlSystemStudio/phoebus/issues/3198#issuecomment-2494607006",
"url": "https://github.com/ControlSystemStudio/phoebus/issues/3198#issuecomment-2494607006",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------xaZjcIhMX4F2J6nFbnOrUo5K--

sifriede, 2024-11-25:
> I think your build still contains a bit too much... With the
> dependencies and exclusions I have defined in a build for your use
> case I have the following logbook related jars in the |lib| directory:
>
> |app-logbook-elog-4.7.4-SNAPSHOT.jar|
> |app-logbook-ui-4.7.4-SNAPSHOT.jar|

Dear Georg,

Thanks to your advice, at least I now get results. Unfortunately they 
are not reliable:

The Log Entry Table does show results, yet ignoring the search string 
(see error messages below), but it displays all messages, while Log 
Entry Calender does not:

It somehow tries do add Scratch Pad, when I scroll through the Table 
multiple times:
2024-11-25 12:53:23 INFO [org.phoebus.framework.jobs] Job 'searching 
logbook for : {search=*}': Finished - Cancelled
2024-11-25 12:53:23 INFO [org.phoebus.logbook.ui.write.LogEntryModel] 
Adding default logbook "Scratch Pad"

When using the advanced search and adding the time scale the following 
error messages appear:

2024-11-25 13:03:25 INFO [org.phoebus.framework.jobs] Job 'searching 
logbook for : {search=*, start=1731931399, end=1732536199}': Finished - 
Cancelled
java.time.format.DateTimeParseException: Text '1731931405' could not be 
parsed at index 0
     at 
java.base/java.time.format.DateTimeFormatter.parseResolved0(DateTimeFormatter.java:2052)
     at 
java.base/java.time.format.DateTimeFormatter.parse(DateTimeFormatter.java:1954)
     at java.base/java.time.LocalDateTime.parse(LocalDateTime.java:494)
     at org.phoebus.elog.api.ElogClient.findLogs(ElogClient.java:279)
     at 
org.phoebus.logbook.ui.LogbookSearchJob.lambda$getRunnable$0(LogbookSearchJob.java:56)
     at 
java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
     at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
     at 
java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
     at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
     at 
java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
     at 
java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
     at java.base/java.lang.Thread.run(Thread.java:840)
Exception in thread "JavaFX Application Thread" 
java.lang.NullPointerException: Cannot invoke "java.util.List.stream()" 
because "this.logEntries" is null
     at 
org.phoebus.logbook.ui.LogEntryCalenderViewController.refresh(LogEntryCalenderViewController.java:283)
     at 
org.phoebus.logbook.ui.LogEntryCalenderViewController.setLogs(LogEntryCalenderViewController.java:278)
     at 
org.phoebus.logbook.ui.LogbookSearchController.lambda$search$0(LogbookSearchController.java:41)
     at 
com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:457)
     at 
java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
     at 
com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:456)
     at 
com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
     at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
     at 
com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:316)
     at java.base/java.lang.Thread.run(Thread.java:840)

Best regards
Simon

P.S.: In the meantime I've tested py_elog from the PSI on which Florian 
has developed his elog api 
[... 862 lines cut ...]
sKkG/YwKrVeL6klB65ZUSkKZZiUPWwspcIROSIseEcgWCtEMMsawUZ5d7Qo0sWeuQ8/Dnrrk
k0/z2DGfi+woyI5JBCQCEgGJgERAIpCfCEg5OP/Q//XXXwWhm5KSkumVK5qTaWSYaVaoRGRR
bFczSV3em4K4UH610Z5WyI68hyxshG52hoHDFLPlM9OqvBk/fuwY6tarJ0hYxTJaqU3NQ6tY
TdvdngW8mXaSSuxqpmeVME5KyGv6RwDKpLE9ZrNdOFeGQa1BJ/L8KBcpfLyI4sYDwXUpzLCI
j80su2DqRSxlEurtf6v9FjWqFpz25ik1Ku65kvjNzgyRZSUCEgGJgERAIiARkAhIBIo6Ahx6
uEaNGo5usoduPZIJ8ioMcEHAk4nO3bt3FwhCNy+xVvtZGAhdIcbZvQTZRlsJ363IkIo8SjOH
iEwbnToieXVEgFrpPXvo2th4mAipQMqni+AKsOqpHOXpJEticY2OvHN1FKJZRJhiwV/vbLz8
77//ijDj+XWwAoL7yn1mMVfFQSu7CkJXEYUF2a0oLdjYWumLyDslBF4XiVfzp5SF82uEC959
rWR0z3NH9YgQRvuqPkYsNlV5kqkUUuapvZxmrgr9jNCwcdhwDqSm6JO0840dD2xEIts4Ny4p
2yg1r/1gXZQSelnUzfVqcqgpXhqZhzbkslTuFLx5JVskEZAISAQkAhKBwoSAlIPzb7T++OMP
ESUsI4OMdbkZGhnGSWbR8HnKPlDhEwW3KPasLCgqn/JessgTutxBRRhU4Dh58iRuv/12BAUp
+U+Es7P9VaFVVVLUDivt2rOGvlGhVfDMcqhka+b+XbxzFBUCrPql8sa+r7d/6KZSIeSrbWHh
XynqSWDlweXyek1Y6fybvvLOEgGJgERAIiARkAhIBCQCEoH8QcBoNIK9E1kGUI/NmzejevXq
TnJC/rQub+6qCnrHjx9H06ZN8+Ymftaal1hr+xkRESEISxaY1fPUqVOile4+r127tp89yNti
qpyqvYtCKDlLe1qSR6SNtcuCDmFflUGFzJg15+Z/FKmKFQpCDs6HQ4S+FTmN7HKxC2uV2d2s
Eq5WdM5MlZQpp3uSibV5rfOhy/KW+YyAYqxvX0f8ymtDzDvFmELrHaGsKGV+Khoz1WSeP1cN
99VvM/U1yjulXpclay9s1x05FEB2/Y/SHOVqe5u8wuVpkuczxvL2EgGJgERAIiAR8IaAa3RU
idbNRUDKwfkrB2/ZsgWVK1cW3rXOhzP3l1Vys5O5vFu1GwML1lJ9z69F2UOXHxyqkMyCfVxc
HEqUKCHO/BJmb+7SlXeTCEgEJAISAYmAREAiIBGQCBRPBFiIPX36tDByLFOmjAME/uzKlSuo
WLFikZQJuN/nzp1DqVKlUKVKlXwd/LzEWtvPxMTEIkPoOjz2NFbcKn0peB2V0GXDZA1hpVpx
uyN0L126JNIL8XyQcnC+Lgl5c4mAREAiIBGQCEgEJAISAYlAniIg5eD8l4OPHDki5K7g4GD7
q8aw1o0/Z+ZHmZZ8TiQuyYZGowkmkvWLPKHLq0O13maC9xiFXTabzeK0UB6erG6u9nA6Wawg
td66KneuhMcRBp/20+7367IonStTy2dZua6D6dIG1QI0i2ewMOJ2Y83sGo4qTx8VsnKJgERA
IiARkAhIBCQCEgGJQMFBgPf+0dHRgtjU5k7kz0+cOIHz58+TUGQsOA2+QS1hwbFChQqoVq2a
8ELOzyMvsdb2c+/evUWY0NV4o6ryIodrVgRdZXg5QpMqlbrx+GMjZzZiSEhIyJc5YeF4

georgweiss, 2024-11-26:
Please make sure you have the following in your settings file for Phoebus:
```org.phoebus.logbook.ui/default_logbook_query=search*````

sifriede, 2024-11-27:
Dear Georg,

I've tried that.

Unfortunately, the calender and table apps are not working reliable. I 
have another elog instance using SSL, i.e. https://..., if I use this 
url in the settings the table does not show anything at all. The 
calender does. But I also have to use the "-clean" option to start 
phoebus. It seems there might be some caching problems in my test 
environment.

And if I can't use the search-String-function properly, this does not 
work either in a production environment. The "send to logbook" function 
works reliably.

Best regards

Message ID: ***@***.***>
--------------nhUc0ORNX8ccpMFHEyHeSD0m
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: 8bit

<!DOCTYPE html>
<html>
  <head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
  </head>
  <body>
    <p><font face="Helvetica, Arial, sans-serif">Dear Georg,</font></p>
    <p><font face="Helvetica, Arial, sans-serif">I've tried that. </font></p>
    <p><font face="Helvetica, Arial, sans-serif">Unfortunately, the
        calender and table apps are not working reliable. I have another
        elog instance using SSL, i.e. https://..., if I use this url in
        the settings the table does not show anything at all. The
        calender does. But I also have to use the "-clean" option to
        start phoebus. It seems there might be some caching problems in
        my test environment.<br>
      </font></p>
    <p><font face="Helvetica, Arial, sans-serif">And if I can't use the
        search-String-function properly, this does not work either in a
        production environment. The "send to logbook" function works
        reliably.</font></p>
    <font face="Helvetica, Arial, sans-serif">Best regards</font>
    <p></p>
    <span
style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
      ID: <span>&lt;ControlSystemStudio/phoebus/issues/3198/2500178613</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span>
    <blockquote type="cite"
***@***.***">
      <script type="application/ld+json">[
{
***@***.***": "http://schema.org",
***@***.***": "EmailMessage",
"potentialAction": {
***@***.***": "ViewAction",
"target": "https://github.com/ControlSystemStudio/phoebus/issues/3198#issuecomment-2500178613",
"url": "https://github.com/ControlSystemStudio/phoebus/issues/3198#issuecomment-2500178613",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
***@***.***": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------nhUc0ORNX8ccpMFHEyHeSD0m--

georgweiss, 2024-12-02:
Hi,

We do not have an elog service running, so I have no possibility debugging your issues. 
I need more information, e.g. a stack trace triggered by a search from the client.

georgweiss, 2024-12-03:
Hi,
So it seems the server accepts a search request from the client (first line), indicating that the client has sent it correctly. 
The other lines indicate that the client is requesting log entries identified by id.
Based on this is looks as client is sending correct requests, but how these are handled by the client I cannot determine from the above.
Is there any stack trace or error log message on the client side?

sifriede, 2024-12-03:
Dear George,
Do you mean client = phoebus instance?
The error messages are (both appear for Calender and Table:
``` 2024-12-03 11:11:08 INFO [org.phoebus.framework.jobs] Job 'searching logbook for : {search*=*}': Finished - Cancelled ```
+ I get a lot of the javafx-error messages, like shown above.

Can I somehow increase the error log level or did you mean something different?

georgweiss, 2024-12-03:
Actually that message says that search is done in the background and then terminates. It's not an error message.
Maybe we could setup a Zoom or Teams meeting to make this conversation a bit more efficient?

sifriede, 2024-12-03:
Yes indeed, that would be great. Unfortunately today and tomorrow I am not available anymore but please feel free to write me sifriede@uni-mainz.de so that we can find a suitable appointment.

ffeldbauer, 2025-05-05:
Hi all,
sorry for not replying earlier, but somehow I did not get a notification from github (or it went to spam?).

I wrote the elog client 5 years ago for version 4.6.4 of phoebus.
Since then, the internal logbook API has been modified which broke the calendar and table views for my elog client.

Unfortunately I don't have much time to work on this project. Phoebus (and EPICS) are part of our detector control system for our hadron physics experiment and I'm mainly working on the DAQ (FPGA and hardware developments) now.

I'm not a Java developer, I usually code in C++ (or VHDL), so it was quite some difficult task for me to implement the client the first time as I didn't know which files I had to touch to add my client.

@kasemir @shroffk is there some documentation for the interfaces a logbook client has to implement?
Or some easy to follow example?

If so, I would try to update the elog client to work with the newest phoebus version.
Unless someone else is already working on this?

Cheers,
Florian

wanglin86769, 2025-05-12:
It seems that a few years ago https://github.com/ControlSystemStudio/phoebus/commit/55838d5b20a548a8bd9e532c1a31f9b6c986ba30 added `search()` method to `LogClient` interface in order to support pagination log request which caused this issue. 

https://github.com/ControlSystemStudio/phoebus/pull/3393 is trying to fix this issue with the following modifications,
1. Add `search()` method to `ElogClient.java`
2. Support pagination log request
3. Support sort by date
4. Fix a typo which which causes Text be sent for both text and subject

Another thing to mention is that the following code in `LogEntryEditorController.java` uses `logEntry.getSource()` to display the existing Text in editing mode, which causes the UI display nothing for Elog client because Elog does not have a Source field. 
```
// When editing an existing entry, set the description to the source of the entry.
descriptionProperty.set(editMode.equals(EditMode.UPDATE_LOG_ENTRY) ?  logEntry.getSource() :
        (logEntry.getDescription() != null ? logEntry.getDescription() : ""));
```

Can the default implementation in `LogEntry.java` return Description instead of null as follows? This way, the existing text can be displayed properly for Elog client in editing mode.
```
public default String getSource(){
        return getDescription();
}
```

closed 2025-05-14 by shroffk