ControlSystemStudio/phoebus #3203: Changing Embedded Display Macros Crashes the Screen

UrbanBobek, 2024-12-02:
Hi all,

In our screens we use a script to change macro values of an embedded display to select between device instances. When we change the `file` or `macros` property of embedded display, the screen often crashes.

This happens most often with many levels of embedding and when changing the property quickly from one value to another. I've tried reproducing this behavior with a minimal example display where a script changes the macros property of an embedded display repeatedly. Pressing the button several times makes the screen crash consistently.
[macro_update_bug.zip](https://github.com/user-attachments/files/17975319/macro_update_bug.zip)

The error log output when the attached script crashes the screen:
```bash
2024-12-02 09:49:59 SEVERE [org.csstudio.display.builder.model] Unknown listener org.csstudio.display.builder.representation.ToolkitRepresentation$$Lambda$812/0x00007f3be0487560@4006f5df
java.lang.Exception: Unknown listener
	at org.csstudio.display.builder.model.properties.PropertyChangeHandler.removePropertyListener(PropertyChangeHandler.java:119)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:398)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.dispose(EmbeddedDisplayRepresentation.java:491)
	at org.csstudio.display.builder.representation.WidgetRepresentation.destroy(WidgetRepresentation.java:88)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeWidget(ToolkitRepresentation.java:435)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeChildren(ToolkitRepresentation.java:410)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:394)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.runtime.ActionUtil.handleClose(ActionUtil.java:205)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.disposeModel(DisplayRuntimeInstance.java:457)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.onClosed(DisplayRuntimeInstance.java:495)
	at org.phoebus.ui.docking.DockItem.handleClosed(DockItem.java:714)
	at org.phoebus.ui.docking.DockItemWithInput.handleClosed(DockItemWithInput.java:487)
	at org.phoebus.ui.docking.DockItem.close(DockItem.java:757)
	at org.phoebus.ui.docking.DockItem.lambda$handleCloseRequest$4(DockItem.java:209)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:457)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:456)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:316)
	at java.base/java.lang.Thread.run(Thread.java:840)

Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
	at java.base/java.util.Objects.requireNonNull(Objects.java:209)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:400)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.dispose(EmbeddedDisplayRepresentation.java:491)
	at org.csstudio.display.builder.representation.WidgetRepresentation.destroy(WidgetRepresentation.java:88)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeWidget(ToolkitRepresentation.java:435)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeChildren(ToolkitRepresentation.java:410)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:394)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.runtime.ActionUtil.handleClose(ActionUtil.java:205)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.disposeModel(DisplayRuntimeInstance.java:457)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.onClosed(DisplayRuntimeInstance.java:495)
	at org.phoebus.ui.docking.DockItem.handleClosed(DockItem.java:714)
	at org.phoebus.ui.docking.DockItemWithInput.handleClosed(DockItemWithInput.java:487)
	at org.phoebus.ui.docking.DockItem.close(DockItem.java:757)
	at org.phoebus.ui.docking.DockItem.lambda$handleCloseRequest$4(DockItem.java:209)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:457)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:456)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:316)
	at java.base/java.lang.Thread.run(Thread.java:840)
2024-12-02 09:50:09 SEVERE [org.csstudio.display.builder.model] Unknown listener org.csstudio.display.builder.representation.ToolkitRepresentation$$Lambda$812/0x00007f3be0487560@4006f5df
java.lang.Exception: Unknown listener
	at org.csstudio.display.builder.model.properties.PropertyChangeHandler.removePropertyListener(PropertyChangeHandler.java:119)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:398)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.dispose(EmbeddedDisplayRepresentation.java:491)
[... 29 lines cut ...]
Caused by: java.lang.NullPointerException
	at java.base/java.util.Objects.requireNonNull(Objects.java:209)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:400)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.dispose(EmbeddedDisplayRepresentation.java:491)
	at org.csstudio.display.builder.representation.WidgetRepresentation.destroy(WidgetRepresentation.java:88)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeWidget(ToolkitRepresentation.java:435)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeChildren(ToolkitRepresentation.java:410)
	at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:394)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
	at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
	at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.lambda$updatePendingDisplay$0(EmbeddedDisplayRepresentation.java:306)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:457)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:456)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:316)
	... 1 more

```

Any help is appreciated!

kasemir, 2024-12-02:
May be related to #3063

UrbanBobek, 2025-01-15:
Thank you both for the reply and I apologize for the late reply from my side.

I was using version `v4.7.3` and I see the fix was implemented after this release. 

Using the current `master` branch seems to have helped. I cannot reproduce the issue on our production screens, however, if I press the button in the example I attached repeatedly, I sometimes get the following warning messages.

```
2025-01-15 10:17:38 WARNING [org.csstudio.display.builder.runtime] Skipping display shutdown for EmbeddedDisplay sub2.bob, state was Starting
``` 

If I increase the amount of times the macros are updated in the script from 100 to 1000 and run the script by pressing the button, I  get different outputs in the error log. 

Most commonly I get
```
2025-01-15 10:48:29 WARNING [org.csstudio.display.builder.representation] Failed to represent embedded display
java.lang.NullPointerException: Cannot invoke "javafx.scene.Parent.getProperties()" because "model_parent" is null
        at org.csstudio.display.builder.runtime.app.DockItemRepresentation.representModel(DockItemRepresentation.java:140)
        at org.csstudio.display.builder.runtime.app.DockItemRepresentation.representModel(DockItemRepresentation.java:32)
        at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.representContent(EmbeddedDisplayRepresentation.java:346)
        at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.lambda$updatePendingDisplay$1(EmbeddedDisplayRepresentation.java:316)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:456)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:455)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:263)
        at java.base/java.lang.Thread.run(Thread.java:840)

```

If I really spam the button, so the script runs several times per second, I get an output very similar to the one I sent in the first message. However, this example does feel a bit extreme. Is this behavior expected? 

```
2025-01-15 10:52:58 SEVERE [org.csstudio.display.builder.model] Unknown listener org.csstudio.display.builder.representation.ToolkitRepresentation$$Lambda$835/0x00007f58e84af9f8@4182e1b4
java.lang.Exception: Unknown listener
        at org.csstudio.display.builder.model.properties.PropertyChangeHandler.removePropertyListener(PropertyChangeHandler.java:119)
        at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:398)
        at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
        at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
        at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.lambda$dispose$2(EmbeddedDisplayRepresentation.java:493)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:456)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:455)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:263)
        at java.base/java.lang.Thread.run(Thread.java:840)

Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
        at java.base/java.util.Objects.requireNonNull(Objects.java:209)
        at org.csstudio.display.builder.representation.ToolkitRepresentation.disposeRepresentation(ToolkitRepresentation.java:400)
        at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:663)
        at org.csstudio.display.builder.representation.javafx.JFXRepresentation.disposeRepresentation(JFXRepresentation.java:115)
        at org.csstudio.display.builder.representation.javafx.widgets.EmbeddedDisplayRepresentation.lambda$dispose$2(EmbeddedDisplayRepresentation.java:493)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:456)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:455)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:263)
        at java.base/java.lang.Thread.run(Thread.java:840)
2025-01-15 10:52:59 WARNING [org.csstudio.display.builder.representation] Failed to represent embedded display
java.lang.NullPointerException: Cannot invoke "javafx.scene.Parent.getProperties()" because "model_parent" is null
[... cut ...]
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:455)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$10(GtkApplication.java:263)
        at java.base/java.lang.Thread.run(Thread.java:840)

```