ControlSystemStudio/phoebus #3204: .OAV in acalcout record respects .PREC, but .AVAL does not

bsobhani, 2024-12-05:
[image]

Notice that 1,1,1... is displayed for the CALC output whereas 1.300,1.300,1.300... is displayed for the OCAL output. I would expect 1.300,1.300,1.300... for both, since .PREC is set to 3.

This is a screenshot of the standard acalcout CSS screen, but I see the same behavior in a plain textupdate widget in a new CSS screen - OAV respects .PREC but .AVAL does not.

kasemir, 2024-12-05:
This seems to be how the acalcout record determines the precision of channels for the various fields.
VAL uses PREC. For the rest, it calls recGblGetPrec, where I'm not sure how that works, maybe you can look into it.

https://github.com/epics-modules/calc/blob/9e99f4e945abb666c5e0d247e35ea780603738e3/calcApp/src/aCalcoutRecord.c#L747

bsobhani, 2024-12-05:
caget -d DBR_CTRL_DOUBLE returned the same values with the same precision for both .OAV and .AVAL

I found that if I restart phoebus (i.e. a full restart of the application, not just refreshing the screen), then upon restart the .AVAL and .OAV fields are both of the precision of the .PREC field. So I think the precision is somehow being cached and doesn't get updated until phoebus restarts.

I guess now that I know the trick it doesn't seem like as big of an issue, but may be still something to think about. I will close the issue.

closed 2024-12-05 by bsobhani

kasemir, 2024-12-05:
Again I'm not sure what recGblGetPrec does. Maybe the precision of some of those channels changes as you configure for example inputs of the record?
If the precision of channels changes at runtime, phoebus will only see those changes when configured with 

    org.phoebus.pv.ca/dbe_property_supported=true

The default is false, so that would then require a restart to see changed display metadata.

bsobhani, 2024-12-05:
I forgot to mention I was manually changing the PREC field. So for my initial problem I had processed AVAL with the default precision. Then I changed the PREC field, and saw the AVAL precision was not changing. Then later I experimented with OCAL and noticed that the precision for OAV was correct, whereas AVAL was still not. I initially thought it was a difference between how CSS treats the AVAL and OAV fields, but really it seems like it was probably some kind of caching.

On my setup org.phoebus.pv.ca/dbe_property_supported was already set to true. Scalar values change to the new PREC field precision upon page refresh, but array values do not. Behavior is exactly the same if I set org.phoebus.pv.ca/dbe_property_supported to false.

[image]