ControlSystemStudio/phoebus #3229: Installing error

PeiyuQuan, 2025-01-10:
Hello
My system us Rocky 9.
When i run this command:
mvn clean install
I got the following error information.

[INFO] Tests run: 5, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.008 s - in org.epics.pva.data.StructureTest
[INFO] Running org.epics.pva.data.SerializeTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 s - in org.epics.pva.data.SerializeTest
[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[ERROR]   ServerClientTest.testSinglePV:262
[INFO] 
[ERROR] Tests run: 63, Failures: 12, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for phoebus (parent) 4.7.4-SNAPSHOT:
[INFO] 
[INFO] phoebus (parent) ................................... SUCCESS [  0.318 s]
[INFO] dependencies ....................................... SUCCESS [  0.014 s]
[INFO] install-jars ....................................... SUCCESS [  0.449 s]
[INFO] phoebus-target ..................................... SUCCESS [  0.573 s]
[INFO] core ............................................... SUCCESS [  0.007 s]
[INFO] core-pva ........................................... FAILURE [02:04 min]
[INFO] core-vtype ......................................... SKIPPED
[INFO] core-formula ....................................... SKIPPED
[INFO] core-framework ..................................... SKIPPED
[INFO] core-types ......................................... SKIPPED
[INFO] core-util .......................................... SKIPPED
[INFO] core-pv ............................................ SKIPPED
[INFO] core-pv-ca ......................................... SKIPPED
[INFO] core-pv-jackie ..................................... SKIPPED
[INFO] core-pv-mqtt ....................................... SKIPPED
[INFO] core-pv-opva ....................................... SKIPPED
[INFO] core-pv-pva ........................................ SKIPPED
[INFO] core-pv-tango ...................................... SKIPPED
[INFO] core-security ...................................... SKIPPED
[INFO] core-ui ............................................ SKIPPED
[INFO] core-logbook ....................................... SKIPPED
[INFO] core-email ......................................... SKIPPED
[INFO] core-launcher ...................................... SKIPPED
[INFO] app ................................................ SKIPPED
[INFO] app-diag ........................................... SKIPPED
[INFO] app-log-configuration .............................. SKIPPED
[INFO] app-probe .......................................... SKIPPED
[INFO] app-pvtable ........................................ SKIPPED
[INFO] app-pvtree ......................................... SKIPPED
[INFO] app-update ......................................... SKIPPED
[INFO] app-email .......................................... SKIPPED
[INFO] app-email-ui ....................................... SKIPPED
[INFO] app-errlog ......................................... SKIPPED
[INFO] app-filebrowser .................................... SKIPPED
[INFO] app-logbook ........................................ SKIPPED
[INFO] app-logbook-inmemory ............................... SKIPPED
[INFO] app-logbook-ui ..................................... SKIPPED
[INFO] app-logbook-olog ................................... SKIPPED
[INFO] app-logbook-olog-client ............................ SKIPPED
[... 60 lines cut ...]
[INFO] service-alarm-logger ............................... SKIPPED
[INFO] service-alarm-config-logger ........................ SKIPPED
[INFO] service-save-and-restore ........................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  02:06 min
[INFO] Finished at: 2025-01-09T19:37:40-08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M7:test (default-test) on project core-pva: There are test failures.
[ERROR] 
[ERROR] Please refer to /home/red/phoebus/core/pva/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <args> -rf :core-pva

Thank you

kasemir, 2025-01-10:
I get that same error,

```
[ERROR] Failures:
[ERROR] ServerClientTest.testSinglePV:262
```

unless I add this to the `testServer()` code:
```
PVASettings.EPICS_PVA_ADDR_LIST = "127.0.0.1";
```

Details can vary based on your network setup.
The unit tests are mostly for those who develop the code.
If you build CS-Studio from source code simply to use it, you can skip the unit tests as described in the [README.md](https://github.com/ControlSystemStudio/phoebus/blob/master/README.md by using `mvn -DskipTests ...`.

PeiyuQuan, 2025-01-10:
Hello Kay

It works. Thank you.

If i just used the command "mvn -DskipTests ...", i got the same error. When i modified ServerClientTest.java according to your code. Phoebus has been installed. Thank you.

        {
            // Details vary with network setup,
            // but forcing use of IPv4 localhost is quite likely to "work"
            PVASettings.EPICS_PVAS_INTF_ADDR_LIST = "127.0.0.1";
            PVASettings.EPICS_PVA_ADDR_LIST = "127.0.0.1";    
        }
The following just show the result from "mvn -DskipTests ..."

(phoebus-community) [red@EPICS1 phoebus]$ mvn -DskipTests ...
[INFO] Scanning for projects...
[WARNING] 
[WARNING] Some problems were encountered while building the effective model for org.phoebus:app-logbook-olog-ui:jar:4.7.4-SNAPSHOT
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: org.phoebus:core-ui:jar -> duplicate declaration of version 4.7.4-SNAPSHOT @ line 43, column 21
[WARNING] 
[WARNING] It is highly recommended to fix these problems because they threaten the stability of your build.
[WARNING] 
[WARNING] For this reason, future Maven versions might no longer support building such malformed projects.
[WARNING] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO] 
[INFO] phoebus (parent)                                                   [pom]
[INFO] dependencies                                                       [pom]
[INFO] install-jars                                                       [jar]
[INFO] phoebus-target                                                     [jar]
[INFO] core                                                               [pom]
[INFO] core-pva                                                           [jar]
[INFO] core-vtype                                                         [jar]
[INFO] core-formula                                                       [jar]
[INFO] core-framework                                                     [jar]
[INFO] core-types                                                         [jar]
[INFO] core-util                                                          [jar]
[INFO] core-pv                                                            [jar]
[INFO] core-pv-ca                                                         [jar]
[INFO] core-pv-jackie                                                     [jar]
[INFO] core-pv-mqtt                                                       [jar]
[INFO] core-pv-opva                                                       [jar]
[INFO] core-pv-pva                                                        [jar]
[INFO] core-pv-tango                                                      [jar]
[INFO] core-security                                                      [jar]
[INFO] core-ui                                                            [jar]
[INFO] core-logbook                                                       [jar]
[INFO] core-email                                                         [jar]
[INFO] core-launcher                                                      [jar]
[INFO] app                                                                [pom]
[INFO] app-diag                                                           [jar]
[INFO] app-log-configuration                                              [jar]
[INFO] app-probe                                                          [jar]
[INFO] app-pvtable                                                        [jar]
[INFO] app-pvtree                                                         [jar]
[INFO] app-update                                                         [jar]
[INFO] app-email                                                          [pom]
[INFO] app-email-ui                                                       [jar]
[INFO] app-errlog                                                         [jar]
[INFO] app-filebrowser                                                    [jar]
[INFO] app-logbook                                                        [pom]
[INFO] app-logbook-inmemory                                               [jar]
[INFO] app-logbook-ui                                                     [jar]
[INFO] app-logbook-olog                                                   [pom]
[INFO] app-logbook-olog-client                                            [jar]
[INFO] app-logbook-olog-client-es                                         [jar]
[INFO] app-logbook-olog-ui                                                [jar]
[INFO] app-logbook-elog                                                   [jar]
[INFO] app-rtplot                                                         [jar]
[... 157 lines cut ...]
[INFO] app-utility-preference-manager ..................... SKIPPED
[INFO] app-imageviewer .................................... SKIPPED
[INFO] app-credentials-management ......................... SKIPPED
[INFO] app-eslog .......................................... SKIPPED
[INFO] product ............................................ SKIPPED
[INFO] services ........................................... SKIPPED
[INFO] service-alarm-server ............................... SKIPPED
[INFO] service-archive-engine ............................. SKIPPED
[INFO] service-scan-server ................................ SKIPPED
[INFO] service-alarm-logger ............................... SKIPPED
[... cut ...] test, integration-test, post-integration-test, verify, install, deploy, pre-site, site, post-site, site-deploy. -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/LifecyclePhaseNotFoundException

kasemir, 2025-01-13:
> If i just used the command "mvn -DskipTests ...", i got the same error.

> (phoebus-community) [red@EPICS1 phoebus]$ mvn -DskipTests ...
> [INFO] phoebus (parent) ................................... FAILURE [ 0.001 s]
> [ERROR] Unknown lifecycle phase "...". You must specify a valid lifecycle phase or a goal 

Don't literally run `mvn -DskipTests ...`. `...` is not a valid goal.
The `...` stands for whatever you had in there, for example:

```
mvn -DskipTests  clean install
```

Anyway, thanks for confirming that the test proceeds with "127.0.0.1"

kasemir, 2025-01-31:
Fixed with #3230

closed 2025-01-31 by kasemir