ControlSystemStudio/phoebus #3235: Add a widget to display the timestamp of a PV

rosstitmarsh, 2025-01-15:
Some users at ISIS have requested the ability to have the timestamp of a PV displayed on a Phoebus screen.

This could be useful to for screens displaying the latest data from the scan of an invasive diagnostic monitor. The user will imediately see date/time of the last scan without having to hover their mouse over an indicator or another work around.

Current ways to acheive this:

- Reprogram the device to send out an extra string PV containing a human readable timestamp. This adds development overhead, is inflexible and means the timestamp format is fixed.
- Use a script to read in `pva://example:pv/timeStamp/secondsPastEpoch`, format it, then write it to a `loc://` PV. This adds the overhead of creating and running scripts and adds complexity.

A native widget would be much simpler for users and creators of displays. It would also make it possible to add a preference for the timestamp format in the Phoebus preferences making all timestamps consistent and more readable.

kasemir, 2025-01-15:
@shroffk  How about a formula function to support for example
 ```
   =timestamp(`SomePVName`)
```

It would turn the timestamp of a VType into a string like `2025-01-18 13:14:15.123456789`

It could support an optional second formatting parameter:
 ```
   =timestamp(`SomePVName`, "YYYY-MM-DD HH:MM")
```
Ideally, the formatting uses some new syntax that's different from anything anybody has ever seen before.
Or be boring and go with the EPICS strftime, https://epics.anl.gov/base/R7-0/7-docs/doxygen/classepics_time.html#ac5223634796ca115443eb3a86c9c7ae3, which is like the common strftime with an addition to handle nanoseconds.

kasemir, 2025-01-15:
The boring option would have the advantage of expanding the EPICS user base from just the cool kids to the general nerd population.

shroffk, 2025-01-15:
`=timestamp(...)`
This would be easy to implement.

> Ideally, the formatting uses some new syntax that's different from anything anybody has ever seen before.

lol...This is depressingly funny.

However, we have two options... do we use the formatters used in Phoebus.
https://github.com/ControlSystemStudio/phoebus/blob/master/core/util/src/main/java/org/phoebus/util/time/TimestampFormats.java

or

strftime

I am leaning towards just using the Phoebus ones since the are used in other formulas too... like the archiver datasource ( aa:// )

kasemir, 2025-01-15:
> two options

Sadly, Googlefight is nothing more than a wikipedia entry, https://en.wikipedia.org/wiki/Googlefight

Still, my gut feeling suggests that python is more popular than EPICS, so that would certainly find a larger audience. I don't mean to say that the popular vote is necessarily the best, but going with the masses it is obviously an easy choice.

kasemir, 2025-01-15:
.. and python has `strftime`

But yes, the TImestmapFormats are nice as well.

shroffk, 2025-01-16:
https://github.com/ControlSystemStudio/phoebus/pull/3239

Currently, you can give any format that DateTimeFormatter accepts

kasemir, 2025-01-31:
Handled by #3239, using Text Update widget with
```
 =timestamp(`NameOfPV`)
```

closed 2025-01-31 by kasemir