ControlSystemStudio/phoebus #3251: Can no longer enter PV names into the PV Table

kasemir, 2025-01-31:
.. this is on a Mac.

Open PV table, click in the "<enter new PV>" region.
[image]

A multi-row text field appears for entering PV names. PV name completion is functional
[image]

But once I entered a name or several names, there's no way to leave the text field.
Enter, shift-enter, control-enter, .. all stay in the text field.
[image]

Pressing Escape or clicking somewhere outside the text field close the text field, and I'm back where I started with an empty PV table.

Not sure which of the recent PV table changes causes this (#3246, #3228, #3237), maybe @katysaintin can take a look?

kasemir, 2025-01-31:
Same on Linux: Can't enter PV names, rendering the PV table pretty useless

kasemir, 2025-01-31:
It is possible to add PV names via drag & drop, but doing that on Linux brings up the d&d issue #3132

katysaintin, 2025-02-03:
@kasemir @shroffk , yes I understand, I will fix this regression ASAP. Katy

katysaintin, 2025-02-03:
Hello , here is the bug fix . I propose to activate the TextArea editor in the preferences and backup to the old behavior by default.
The idea, is to copy/past a list of PV in the PVTable to create a PV Table very quickly. 
Now, we can activate the feature in the preferences, also I don't find a way to map "Enter" action in TextArea to commit the modifications, because Enter, is used to return to line . That's why on right click , I override the Context Menu to add the PV List.
I hope it is OK like that.

Katy

katysaintin, 2025-02-03:
I found a solution to commit the value on ENTER key event on a TextArea also. But still let the TextField as the default editor.

kasemir, 2025-02-03:
Good, this has been fixed, I can again enter PV names.

But now it somehow changes the order of PV names.

Enter PVs a, b, c, d and they turn into what looks like a random order:

[image]

kasemir, 2025-02-03:
.. and when I try to enter a comment like `# This is a comment`, that turns into separate PVs "This", "is", "a", "comment"?!

kasemir, 2025-02-03:
In the past, we could use the PV table to restore settings, in a specific order, awaiting completion on each item, with comments like this:

[image]

Now the order of PVs is changed and I can no longer enter comments...

kasemir, 2025-02-03:
I can also no longer enter PV names like `loc://text("This is text")` because that gets chopped into separate PVs.

[image]

katysaintin, 2025-02-03:
> I can also no longer enter PV names like `loc://text("This is text")` because that gets chopped into separate PVs.

Hello @kasemir , I check that .

shroffk, 2025-02-03:
Thanks @katysaintin for taking on these bugs

I will hold off on the 5 release until we have a resolution
- [ ] preserve order
The support for spaces 
- [ ] allow comments ( with spaces )
- [ ] allow pv names with spaces ( like the loc pv example )

katysaintin, 2025-02-03:
I'm working on the order now, normally this is the last issue .

katysaintin, 2025-02-03:
Hello @shroffk & @kasemir . Normally it's all OK. The last one about order was due to the commit on Filtering feature. It has to be initialized with the same comparator (SORT_NEW_ITEM_LAST) of SortedList .

kasemir, 2025-02-03:
Good, let's merge that

kasemir, 2025-02-03:
All good now, thanks!

commit 69391fa1 (2025-02-03): Commit value on Enter KeyEvent and add a new line on ShiftDown + Enter
fix #3251
closed 2025-02-03 by kasemir