ControlSystemStudio/phoebus #3276: Change in EmbededDisplayRepresentation affects existing OPIs

georgweiss, 2025-02-14:
Changes in #3193 related to ```EmbeddedDisplayRepresentation``` apparently have an effect on the width (and height?) of an Embedded Display. In an OPI where Embedded Displays are pretty narrow and stacked horizontally, the width available for the widgets it contains may become too small. This in turn will trigger rendering of vertical and horizontal scrollbars, see example below.

@DenSinH, can you please investigate?

In the meanwhile I will create a PR reverting the changes in ```EmbeddedDisplayRepresentation```.

georgweiss, 2025-02-14:
Sadly no, bob file in screen shot is from our production environment, so exporting it would be tricky.
My five cents: adding a ```Group``` container will probably boost the overall width/height of the scen graph. Nor sure how to solve this... Maybe check how the CSS style can be manipulated to make the ```Group``` less "intrusive".

DenSinH, 2025-02-18:
I've tried making a parent display with some horizontally stacked, high and narrow embedded displays, but as long as I set the size of the embedded display to be the same as the embedded display widget's size, I am not getting scrollbars... Can you maybe create a small example for me to test this with?

georgweiss, 2025-02-18:
So I took a closer look at the OPI I've attached a screenshot of. There are several ```TextUpdate``` widgets where user has defined a 3px border, and from what I can determine it is not included in the dimensions of the widget. I'm guessing that when you wrap the embedded display in a JavaFX ```Group```, its default styling may additional size to the overall width/height.

I think my users have adjusted the containing OPI and its embedded displays to be as narrow as possible, but the contribution from the ```Group``` node apparently made the embedded display wide enough to trigger scroll bars.

I've attached a simple example that should demonstrate the issue.

[Archive 2.zip](https://github.com/user-attachments/files/18846156/Archive.2.zip)

DenSinH, 2025-02-19:
Thanks, I'm looking at it at the moment, it seems like the vertical bar shows up because of the horizontal bar, which indeed shows up because of the border. I didn't have much time to look at it further today, it's very strange that the Group somehow changes the layout...