ControlSystemStudio/phoebus #3278: Save and restore Elasticsearch issue

jacob720, 2025-02-17:
Hi,

I've installed Phoebus and want to set up the save-and-restore service. I've followed the instructions in I've followed the instructions from [here](https://github.com/ControlSystemStudio/phoebus/blob/master/services/save-and-restore/README.md)](https://github.com/ControlSystemStudio/phoebus/tree/master/services/save-and-restore), but when I start up the service Elasticsearch seems to fail. After starting the service with `java -jar target/service-save-and-restore-4.7.4-SNAPSHOT.jar`, when I click Applications -> Utility -> Save And Restore, a panel comes up saying 'Save-and-restore service unavailable!'.

When I do `curl --fail-with-body http://localhost:8080/save-restore` nothing comes up, and in my browser at http://localhost:8080/save-restore/ the following text is displayed: `{ "name" : "service-save-and-restore", "version" : "4.7.4-SNAPSHOT", "elastic" : { "status" : "Failed to connect to elastic Connection refused" }, "rootNodeID" : "44bef5de-e8e6-4014-af37-b8f6c8a939a2" }`. 

I was hoping someone would be able to give me some advice or point me in the right direction.

Many thanks

jacob720, 2025-02-19:
Thank you! Managed to get it working

georgweiss, 2025-02-19:
@jacob720, can we close this?

jacob720, 2025-02-19:
Yes sorry, I'll close this now

closed 2025-02-19 by jacob720