ControlSystemStudio/phoebus #3288: [Docs] Dead links in Widget Script Section

eddybl, 2025-02-24:
On the [Access Widget in Script](https://control-system-studio.readthedocs.io/en/latest/app/display/editor/doc/access_widget.html#access-widget-in-script) page, the two links on the bottom lead both to 404 pages and I was not able to figure out what the correct links actually should be

kasemir, 2025-03-06:
When building the product, javadoc needs to run to create those pages, which are then included in the online help.

#3313 has screenshots of Help/About for the SNS product (https://controlssoftware.sns.ornl.gov/css_phoebus) that include those API docs. For read-the-docs, only the `*.rst` pages are used, no javadoc is created so those links fail.

eddybl, 2025-03-10:
Thanks for the clarification.

> For read-the-docs, only the *.rst pages are used, no javadoc is created so those links fail.

Is this a question of "javadoc is currently not implemented but could/should be at some point" or is this something which will more likely not change? We currently use the online docs instead of building our own version, but this might be a reason to change that (or instead work on possibly including that into the online docs)

kasemir, 2025-03-10:
It's a matter of correctly building the help.

The Sphinx `*.rst` format is generic markup that can in principle be rendered in various ways: PDF, html, ...
If we render it as html, we can fundamentally include links to other html docs, like the javadoc-converted-to-html for key widget-related API. But that's somewhat of a hack. If you render the `*.rst` as PDF, then those links won't work. Even when rendered as html, sphinx has no clue about that external html content. You have to create the external content (javadoc -> html) beforehand, then run sphinx.

```
# Create Javadoc
( cd app/display/editor;  ant -f javadoc.xml clean all )

# Create documentation
( cd docs; make clean html )
```

The first step obviously creates the javadoc.
The second step runs sphinx to render all the *.rst.
Our sphinx `conf.py` has been tweaked to copy the pre-existing html files into the output, see
https://github.com/ControlSystemStudio/phoebus/blob/0ae9b167491f6f33bea46f998e58988b3e20e34d/docs/source/conf.py#L275C1-L276C1

When we build the online help as part of building CS-Studio, we can do this (javadoc, then sphinx).
When we use the free read-the-docs mechanism to publish the help online, it's doing a good job at running sphinx, but it doesn't do the initial javadoc step.