ControlSystemStudio/phoebus #3313: Can I get a list of macros that bob file sees in python script

hinxx, 2025-03-05:
I want to access macros inside the embedded python script that are defined (inherited) in the bob file. Is there a way to do this?

Should this method be used?
https://github.com/ControlSystemStudio/phoebus/blob/af20f73a7840a2c33d518e243373c4b0951e22aa/app/display/runtime/src/main/java/org/csstudio/display/builder/runtime/script/DataUtil.java#L78C25-L78C38

What is the purpose of `boolean include_parent_macros` and how is it used here?

kasemir, 2025-03-05:
```
# Expand specific macro
widget.getEffectiveMacros().getValue("MacroName")

# List all macros
widget.getEffectiveMacros().getNames()

# Expand all macros in a string
MacroHandler.replace(widget.getEffectiveMacros(), "$(NAME) $(WHATEVER)")
```

See https://github.com/ControlSystemStudio/phoebus/blob/af20f73a7840a2c33d518e243373c4b0951e22aa/app/display/model/src/main/java/org/csstudio/display/builder/model/Widget.java
and
https://github.com/ControlSystemStudio/phoebus/tree/af20f73a7840a2c33d518e243373c4b0951e22aa/core/framework/src/main/java/org/phoebus/framework/macros

The `DataUtil` with `include_parent_macros` was added for compatibility with older BOY scripts. It's no longer applicable.
 Macros are _always_ inherited.

hinxx, 2025-03-06:
Thanks! `widget.getEffectiveMacros().getValue("MacroName") ` is what I was looking for. Did I miss that in the docs somehow?

kasemir, 2025-03-06:
It's in the bundled About/help:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2025-03-06 by kasemir